hbitter/DNMS #2: Teensy 4.0 und MEMS Alternativen

domcross, 2019-12-20:
Hallo,

Die Firmware für Teensy 4.0 in ja noch in der Entwicklung. Ist schon ein grober Zeithorizont absehbar, wann die Produktionsreife erreicht sein wird?

Das verwendete Mikrofon InvenSense ICS-43434, insbesondere die Variante mit Breakoutboard von Pesky ist in Deutschland schwer zu beschaffen. Gibt es besondere Gründe für genau dieses Modell?
Wie sieht es mit Alternativen aus, z.b. dem Knowles SPH0645 - das gibt es mit Breakoutboard z.B. von Adafruit (Produkt Nr. 3421)? Auf dem Datenblatt sind sich die beiden Mikrofone sehr ähnlich.

Viele Grüße
Dominik

Nordrunner01, 2019-12-20:
@domcross 

Jupp, sehe ich auch so. Ich schätze zwar die Arbeit von Pesky, aber.......... Allein der Versand. :-/

Ich werde mein Glück mit einem Günstigprodukt vom freundlichen Ali versuchen (so um die 3€ronen, inkl. P&V). Soviel ist da ja nicht drann. Eigentlich ein Breakoutboard.

Gruß

glosair, 2020-02-26:
I have designed a board to hold a ICS-43434, I2C line driver and the Teensy 4.0 (Same foot print as the Teensy 4.0)

I bought 10 ICS-43434 for just over £20 delivered from Farnell here in the UK.

Tim

[image]

glosair, 2020-03-22:
I've also an ESP12F board designed ready to go....    

I'll make the desgignes open source when fully tested.

[image]

hbitter, 2020-05-04:
Firmware for Teensy 4.0 is now available.

domcross, 2020-06-02:
@hbitter Bitte noch einen Kommentar zu den Alternativen für das MEMS-Mikro ICS-43434, z.B. dem Knowles SPH0645 - das gibt es mit Breakoutboard z.B. von Adafruit (Produkt Nr. 3421)? Auf dem Datenblatt sind sich die beiden Mikrofone sehr ähnlich - könnte man das verwenden?

closed 2020-07-16 by hbitter