hbitter/DNMS #3: Möglicher Platinenlieferrant ?

Nordrunner01, 2019-12-20:
Moin Herr @hbitter .

Soweit ich gesehen habe, stellen Sie freundlicher Weise die Platinendaten zur Verfügung, aber eine Möglichkeit diese fertig (unbestückt) zu beziehen ist leider nicht vorhanden.

Darf ich dazu etwas vorschlagen?

Es gibt  den Lieferranten [PCBs.io](https://pcbs.io/) . Ich habe vor längerer Zeit mal für ein Projekt dort Platinen eingestellt und diese für die Allgemeinheit zugänglich gemacht. Dies gescha in Absprache mit dem Ersteller.

Soweit ich mich erinnern kann, müssen die Daten in einem Format vorliegen, wie sie EagleCAD erstellt (siehe Punkt PCB Specifications). Ich hatte mir damals eine Trialversion installiert, da PCBs.io die original Files nicht akzeptieren wollte.

Desweiteren finde ich das Konzept gut, welches eine Gutschrift vorsieht für den eigenen Bedarf an Platinen (siehe Punkt *Rewards).

Vielleicht sehen sie das auch so und stellen ihre Werke dort ein und Teilen den Link dahin. So können sich interessierte Nutzer ihr Platinen bestellen.

Ach ja,....... , Hätte ich fast vergessen. Hier mal der Link zu dem was ich damals umgesetzt habe: 
https://pcbs.io/share/r1Dq6

Gruß

Franz4596, 2020-03-12:
Hallo Nordrunner01, 
ich habe einige DNMS BauTeile verfügbar, u.a. mehrere PCB's "DNMS T3.6 V1.2" und "AIRROHR V1.2" gekauft. Würde Platinen auch zu Selbstkosten (~2€) abgeben. 
mfg franz

Nordrunner01, 2020-03-12:
@Franz4596 
Moinsen.

Vielen Dank für dein Angebot. Mein Bedarf ist allerdings gerade gedeckt. Ich bin über einen User schon versorgt.
Siehe: https://forum.iot-usergroup.de/t/umgebungslaerm-erfassen/552
und: https://forum.iot-usergroup.de/t/nordrunner-s-klueterkammer/130/35 .
Wir haben im Forum auch einen Bereich für suche/biete. Ich würde mich freuen, wenn du deine überzähligen Platinen auch dort anbieten würdest.
Btw., wo hast du fertigen lassen? pcbs.io scheint iwie unzuverlässig zu sein.

Gruß

Franz4596, 2020-03-12:
Ich bestelle PCBs gerne bei aisler.net, weil Lieferzeit 4-8 Tage aus BRD aber min. x3 Stk., ~8€/Stk.
Günstiger ist cn (~1€...3€/Stk) dauert ztw. sehr lange (z.B. multitech-pcb.com, ilcpcb.com), und ggf. liegen die Teile Wochenlang beim Zoll. (z.B. Tindie 5x ICS43434 lagen 32 Tage in Frankfurt, +19% USt). Also einzelne Teile Platinen,Stecker oder Vergussmassen zu kaufen ist rel. teuer.
Um aber das DNMS Projekt zu unterstützen, biete ich in soweit Hilfe an, dass ich einzelnen Leuten Teile (leer/bestückte PCB,  fertige Mikros) auf Anfrage (franz.hoefle[AT]buergerforum-gladbeck.de) anbiete, ich möchte aber kein Geschäft daraus machen.

Nordrunner01, 2020-03-13:
> Ich bestelle PCBs gerne bei aisler.net, weil Lieferzeit 4-8 Tage aus BRD aber min. x3 Stk., ~8€/Stk.
> Günstiger ist cn (~1€...3€/Stk) dauert ztw. sehr lange (z.B. multitech-pcb.com, ilcpcb.com), und [...]

Rrrrrrespekt! Schönes Ding!
Sowas mag ich!
@Franz4596 , sach ma, was soll das Mikro von Pesky bei Dir kosten inkl. P&V? Ich melde hier mal Interesse an und möchte auf eines den Daumen drauf legen.

Gruß

capitaenz, 2020-05-17:
Moin zusammen,
ich habe auch Interesse an dem Projekt und habe schon begonnen zu schauen wo ich alles geliefert bekomme. Leider schreckt mich genau dies ab, denn 1. die Versandkosten und die Vielzahl an Versandkosten bei unterschiedlichen Händlern und dann 2. die Verfügbarkeit des Mikrofons.
Gibt es mittlerweile irgendwo die Möglichkeit **ein Set zu erwerben**? Das PU Gießharz und die Rohre sehe ich jetzt nicht als Problem selber zu organisieren.

VG
Leif

hbitter, 2020-05-17:
Hallo Leif,

- komplette Sets gib es z.Zt. noch nicht

- ich habe gerade einige Mikros vergossen und könnte da sicher eines entbehren. Welche Boards benötigst Du denn T3.6 o. T4.0, Airrohr? Schick mir bitte eine Mail was Du benötigst (helmut.bitter@t-online.de)
Viele Grüße
Helmut

unrzn0, 2020-08-17:
Hab gerade das hier entdeckt, scheint auch mit Hilfe von @hbitter entstanden zu sein: https://github.com/OpenAirCgn/openmike - erscheint mir sinnvoller als das Pesky Board. Das Pesky Board lässt sich auch via OSH Park bestellen - dann sind es nur 60ct für 3 Stück - bestücken muss man dann eben selbst. Das Openmike kostet 95ct/3Stück. Ich hab gerade mal beide bestellt - aber wegen Mittelkürzung beim USPS wird's wohl bis nach der Präsidentenwahl dauern bis da was kommt ...

Franz4596, 2020-10-31:
Gebe einzelne DNMS/Airrohr-Platinen-V1.4 für 0,60€/Stk+Porto u. ggf. Mikro an Bastler ab, <franz.hoefle(a)online.de>.