hbitter/DNMS #4: problem compilation with DNMS 1.1.0

papybricol, 2020-01-05:
Hello,
I tried to test the DNMS with a Teensy 3.6 using the arduino IDE (vers. 1.8.9).
With the firmware "DNMS_1.1.0.ino" and the library "wire" (from Teensy) it was modified (buffer 64).
The "i2c_t3" library is present in the Teensy folder.

Compilation fails: see attached error messages.

[messages d'erreur.docx](https://github.com/hbitter/DNMS/files/4023541/messages.d.erreur.docx)
There would be: "multiple definition of: i2c0_isr, i2c1_isr" ... in WireKinetis.h ??

Thanks

atze09, 2020-02-27:
> Hello,
> I tried to test the DNMS with a Teensy 3.6 using the arduino IDE (vers. 1.8.9).
> With the firmware "DNMS_1.1.0.ino" and the library "wire" (from Teensy) it was modified (buffer 64).
> The [...]

Did you already solve the Problem? I would be interested in the solution.

papybricol, 2020-02-28:
No, I gave up. It was to test the Teensy with the IDE.
The sensor is now in operation as in the luftdaten assembly

hbitter, 2020-02-29:
@papybricol 

I send you a mail with the following content  and don' t know id you didn't get it or if it did not work for you:

If you like to compile the sketch yourself, you have to exchange the includes of "Wire.h" to "i2c_t3.h"  in the Teensy 3.6 Audio Library.

First you have to find the Teensy Audio Library on your system.
The next pictures shows it on my system and the highlighted modules have to be changed but the the .cpp files and not the .h files:
[image]

Next picture shows the change in one of the files:
[image]

I hope, it will compile than.
Please let me know if it works:  helmut.bitter@t-online.de

Regards,
Helmut

papybricol, 2020-03-06:
@hbitter

Good evening,
I installed the teensy with ESP and everything works fine. The sensor is on the map (1st in France!)
My question was just for a little test, I was waiting for the delivery of the ESP, now I'm not going to disassemble the sensor.
Thank you for your interest
Papybricol

closed 2020-05-06 by hbitter