hbitter/DNMS #5: MEMS günstig beziehen

jensens, 2020-01-17:
Tindie:
> 10,74 € EUR
> Shipping to Austria starts at $14.95 (13,44 € EUR) 

Uhh, Ich mein: 13,44 Versand!? Echt?

Was gibt es sonst noch für sinnvolle Bezugsmöglichkeiten? 

Den nackten ICS43434 gibts bei Ali, für 2,63 und 3,63 Versand (bleibt gleich bei mehreren).
Aber das wird fummeliges Löten. Würde ich gerne vermeiden.
https://de.aliexpress.com/item/4000540560022.html

Nordrunner01, 2020-01-18:
@jensens Moin Namensvetter.

Jupp, es scheint so, als wenn das Produkt von Pesky im Moment die einzige Wahl ist. Ich finde die hohen Versandkosten auch....... :-1: 

Alternative Bezugsmöglichkeiten müssten wir selbst schaffen.

Gruß

Franz4596, 2020-03-21:
Ich habe 02/2020 10x ICS43434 Pesky breakouts gekauft, die lagen fast 4 Wochen bei Zoll Frankfurt und da der Betrag über 22,-€ lag, musste ich 19% USt (ab150€ Zoll) zahlen. 
Der Link Aliexpress für " ICS-43434 MIC" ist nun [https://de.aliexpress.com/item/32918814955.html](url),  24€/10St. Mouser.de liefert das schneller (2,14+USt+Vesand). Dazu kommt noch das beakout, R,C-SMD und smd-löten, rentiert sich nur in größeren Stückzahlen. Da finde ich 15€ akzeptabel.

glosair, 2021-01-02:
I'm currently working on a board that will piggyback under the teensy, first version is too thick and has problems with attaching the ICS43434 so might have to go to a flex pcb 

[image]

hbitter, 2021-01-03:
Hi glosair,
I don' t know what your motivation is to place the microphone on a piggypack underneath the Teensy. One thing you should consider is, that the overall frequency response depends heavily on the casing of the microphone. I think the piggyback board will result in some additional refections that may influence the frequency response and thderefor the measurement results. Such a configuration may need additional adjustment of the filters.
Regards,
Helmut

glosair, 2021-01-03:
Hi Helmut

I'm currently working on a new version that will have the same electronics but build on a flex pcb with the ICS43434 extended away from the teensy on a "ribbon".   The prototype had the port pointing away from the teensy. 

https://invensense.tdk.com/download-pdf/an-1124-recommendations-for-sealing-invensense-bottom-port-mems-microphones-from-dust-and-liquid-ingress/

hbitter, 2021-01-03:
glosair,

that sound interesting. I know that application note from InvenSense. We tried to get hold of the membrame (McMaster-Carr Part Number 86435K31) they suggested in the AN but had no luck so far to get it deliverd to germany. Perhaps you have better connections and could get some piece of it. Definitely I' m interested to get a piece to test it. You can send me a PM helmut.bitter(at)t-online.de
Regards,
Helmut

pe0ter, 2021-01-03:
Good evening Helmut, glosair,

Instead of trying to source a silicone membrane, isn’t it more practical to glue a 3D printed 1mm thick 12.5mm wide ring with a hole of say 10 mm on top of the DNMS microphone unit, then pull the thinnest condome on the market over it, hold it in place with some tape and cover it with the foam cap?

Met vriendelijke groet,
Peter

glosair, 2021-01-03:
Evening Peter

Think the problem with a condom is the uniformity, they generally are formed by dipping so even the thickness of the material changes from one to the next and so would the acoustic properties.  They are also not rated for outdoor use. 

The McMaster-Carr part number appear to be wrong and should be 86435K121 the digits following the K are just the thickness and panel size. 121 is 0.01" thick and 6"x6"

    Color: 10A sheets: Gray; 20A sheets: Black; 35A sheets: Cream
    Temperature Range: -80° to 425° F
    Tensile Strength: 10A sheets: 250 psi; all other sheets: 800 psi
    For Use Outdoors: Yes

Tim

Franz4596, 2021-01-04:
Helmut B. and me, and some friends of me tried to order the **MCMaster-Carr silicone sheet**, with anwer: „Thank you for considering McMaster-Carr for your purchase.  However, we will not accept orders from your location due to the cost of complying with US export regulations.“ and "We have no locations outside of the United States, and cannot suggest another source for these items.  We apologize for not helping you further."

I use for protection a kitchen clingfilm (**Frischhaltefolie**) which is very thin and better than a condome. I put it directly over the top of the micro and measured ~1dB damping. 

And the German Wacker-Siliconfolie (Elastosil / SILPURAN® FILM 2030) shall not good for outdoor applications, due to strong pollute themselve. 

Franz

pe0ter, 2021-01-04:
Danke Franz,

Frischhaltefolie, I have plenty of that.
I have 5 sets DNMS + microphone available, I wil use my iphone as a signal generator and see if I can compare and log the differences between 2 microphones, first without then one with Folie.

Met vriendelijke groet, mit freundlichem Gruß, best regards,
Peter

buv, 2021-01-21:
Hei,

Frischhaltefolie sounds great and is simple. I like that. The foil will act like a drum. I suspect, that depending on the strength one uses to apply the foil to the mic, there will be a resonance at a certain frequency induced in the system. Did anyone try to measure the frequency dependance of the damping induced by the foil?

Greetings,
Burkhard

buv, 2021-01-21:
Hei Helmut,

thank you. Very interesting. So I was roughly wrong by a factor of two with my estimation of the resonance frequency. Talking about response functions: How did you obtain the coefficients of the biquad filter? Does it contain the uncovered mic response?

Greetings,
Burkhard

hbitter, 2021-01-22:
Hi Burkhard,

coefficients of biquad filter:
that was a mixture of using software for calculating the coefficients than doing a lot of measurements and repetitions of that.

We did overall measurements including the microphone frequency response in the proposed casing (1/2" tube and molded).

Regards,
Helmut