hbitter/DNMS #6: Firmware Teensy 4.0

alexvr6, 2020-01-21:
Hallo, 

kann man mittlerweile abschätzen wie lange es mit der Firmware noch dauert? Da auf Luftdaten.info leider nicht erwähnt wird, das die Firmware noch nicht erhältlich ist habe ich mir den Teensy 4.0 bestellt. Habe dann bei Zusammenbau erst mitbekommen, das dir Firmware noch nicht erhältlich ist. 
Ansonsten gute Arbeit, macht weiter so....

julianw10, 2020-02-16:
Genau dasselbe Problem habe ich auch. Da ich dachte, dass der Teensy 4.0 unterstützt wird, wie auf der Website größtenteils, habe ich den auch bestellt. Jetzt muss ich feststellen, dass es keine Firmware gibt. Gibt es ein voraussichtliches Veröffentlichungsdatum oder einen Vorabzugriff auf die Firmware?

mdietinger, 2020-02-20:
Schließe mich den vor Postern an.
Habe mir auch einen Teensy 4.0 zugelegt.
Bin auch gerne bereit Beta Test zu unterstützen.

atze09, 2020-02-21:
Ich mich auch, habe auch bereits den Teensy 4.0 bestellt und würde auch die Beta Phase zu unterstützen!

alexvr6, 2020-02-22:
Da bin ich ja nicht der einzigste...

Wenn jemand interesse hat, ich habe noch 4 Platinen T3.6+NodeMCU V1.0 und eine komplette Bestückte T4+NodeMCU V1.2...

atze09, 2020-02-23:
Eine ungefähre Aussage wäre super. Ob 1 Woche, 1 Monat oder eher 1 Jahr? Dann würde ich auf das Angebot oben zurück kommen;)

atze09, 2020-02-23:
> Da bin ich ja nicht der einzigste...
> 
> Wenn jemand interesse hat, ich habe noch 4 Platinen T3.6+NodeMCU V1.0 und eine komplett Bestückte T4+NodeMCU V1.2...

Ich hätte Interesse an der T3.6 er Platine.

alexvr6, 2020-02-23:
Schreib mir mal zimmerplatz@alexvr6.de

Nordrunner01, 2020-02-25:
Moinsen.

Vielleicht ist ein gangbarer Weg mal bei den Leuten von [luftdaten.info](https://luftdaten.info/) nachzufragen. Immerhin wird dort auf das DNMS verwiesen und die dortige Infrastruktur genutzt. Siehe: https://luftdaten.info/einfuehrung-zum-laermsensor/#top . Fragen zum Projekt werden im Allgemeinen im Git diskutiert: https://github.com/opendata-stuttgart/sensors-software/issues

Gruß

glosair, 2020-02-26:
Didn't realize that the firmware isn't ready yet so designed a board to hold the mic and a teensy 4.0 (the same size as a teensy 4.0) and a new board that holds a high efficiency switching power supply, ESP12F, I2c line driver and RFM96 Lora radio.

domcross, 2020-04-22:
As I understand from #9 the major roadblock is the unstable I2C communication. I just ran over https://github.com/Richard-Gemmell/teensy4_i2c which is a drop in replacement for the Teensy-wire library - did you try that already?

hbitter, 2020-04-23:
Hi Dominik,
the lib from Richard Gemmell is at thge moment the only I²C Arduino lib that supports slave mode. So yes, I tried that lib, with the result, that the I²C communication is very unstable. I tried to conntact Richard Gemmell but so far I got no response. So I hope that the Teensy developer, Paul Stoffregen, will add slave mode to his lib. 
Regards,
Helmut Bitter

closed 2020-05-06 by hbitter