hbitter/DNMS #8: Sensirion SPS30

alexvr6, 2020-02-18:
Da ja das Teensy 4.0 noch nicht funktioniert, habe ich mir DNMS - T3.6+NodeMCU zusammengebastelt. Hier wird jedoch nicht der Sensirion SPS30 ausgelesen. Der Feinstaubsensor funktioniert an sich, nur passiert nichts wenn er an das DNMS - T3.6+NodeMCU angeschlossen ist.
Hat jemand eine Ahnung warum der nicht funktioniert?

mdietinger, 2020-02-18:
Der SPS30 hängt am I2C Bus des NodeMCUs?(D3/D4)
Funktioniert der Sensor wenn T3.6 nicht am Bus hängt?

hbitter, 2020-02-18:
Die NodeMCU Firmware "NRZ-2020-129" hat ein fehlerhaftes I²C Timing. Einige Sensoren funktionieren damit trotzdem (z.B. DNMS oder auch BME280) aber der Sensirion SPS30 Feinstaubsensor nicht!
Bitte die Beta Firmware (z.Zt. "NRZ-2020-130-B5") benutzen, damit funtioniert es zumindest bei mir.

atze09, 2020-03-01:
Hello guys,
First of all this is an awesome project.

I build 3x DNMS - T4+NodeMCU and 1x DNMS - T3.6+NodeMCU and have two new Sensiron SPS30.
Everything work fine, except the Feinstaub sensor Sensiron SPS30.
I tried all the 4 boards and two Sensors on each board and always get the followed error :
`13:23:04.564 -> Read SPS30...
13:23:04.564 -> Trying SPS30 sensor on 0x69H  ... not found
13:23:04.700 -> Check SPS30 wiring`

I measured the voltages of the sensor cable and they where at Pin 1=5V and Pin 2 and 3=3,3V.
Can anybody help me with this issue?
Maybe my sensors are defective..

Thank you
Danke.

Franz4596, 2020-03-21:
SPS30 läuft dank Tipp Beta  "NRZ-2020-130-B5" auch bei mir (mit NRZ-2020-129 nicht).

closed 2020-05-06 by hbitter