hbitter/DNMS #9: Progress of FFT integration

atze09, 2020-04-09:
Hello guys,
i really like your awesome projekt and wanted to ask for the milestone of FFT integration.
Is there a roadmap with any rough estimation for this integration?
Sorry for opening a new issue, but i don't know where to ask for this?
Thank you in advance.
Best Regards

hbitter, 2020-04-10:
Hello atze09,
at the moment I'm still struggling with Teensy4.0 - the I²C communication is still a problem. FFT intergration has to wait until this problem is solved. Basic tools for FFT intergartion are available in the Teensy audio library, nevertheless some measurements have to be done. So timeframe for FFT integration will be 3 - 4 months after Teensy4.0 I²C Comunication is solved.
Regards,
Helmut Bitter
helmut.bitter@t-online.de

atze09, 2020-04-10:
Thank you for your Update.
I will keep the fingers crossed. That the teensy 4 issue will be solved soon.
Best regards,
Dominik

atze09, 2020-09-23:
> Hello atze09,
> at the moment I'm still struggling with Teensy4.0 - the I²C communication is still a problem. FFT intergration has to wait until this problem is solved. Basic tools for FFT intergartion are available in the Teensy audio library, nevertheless some measurements have to be done. So timeframe for FFT integration will be 3 - 4 months after Teensy4.0 I²C Comunication is solved.
> Regards,
> Helmut Bitter
> [helmut.bitter@t-online.de](mailto:helmut.bitter@t-online.de)

Hello Helmut,

Congratulation and many thanks for the Teensy.4.0 integration.
Could you please give us an update of the next steps (FFT, LCeq integration)?
I think many people are interested in the fft frequency analysis to get the root cause/evidence of the main noise disturbtion (Train, truck, ventilation, metal construction, construction noise).
Is the timeframe for the FFT integration still valid?

Best regards,
Dominik

hbitter, 2020-09-23:
Hi Dominik,

it will take a little longer. I will give an update of the next steps and the timeframe in the next days in the READ.me. I think this is interesting for a lot of people.

Best Regards,
Helmut

atze09, 2020-09-23:
> Hi Dominik,
> 
> it will take a little longer. I will give an update of the next steps and the timeframe in the next days in the READ.me. I think this is interesting for a lot of people.
> 
> Best Regards,
> Helmut

Sounds great, thank you for your Feedback.

closed 2020-09-23 by atze09