hbitter/DNMS #10: Wrong Gerber-Files

imaluschi, 2020-06-29:
Hi, i ordered PCB's according to the downloaded DNMS-master.zip from 11.03.2020.
The Gerber Files from DNMS-T3.6-V1.2 where used. Now i've got all parts her and saw, there are only 5 connections for I2C-extension, not 6 for the RJ12-Jack as it was proposed to be.
I saw, on 8to of May there where the gerber files changed, so the actual V1.2 seems different. AAAArgh!!!!

15 PCB's - so how can i "repair" them to use them?
The ordered PCB's, "old" V1.2:
[image]
The "new" V1.2:
[image]

Franz4596, 2020-06-30:
Hi, that is not a problem. 
Cut off one RJ12-Connector from RJ12-6p6s-cable and solder the 6 wires to J2 dirctly, or use a 4 pin female pin header, see photo.
[image]
You can drill RJ12 holes in PCB and solder Cu wires between RJ12 and J2. 
greetings franz

imaluschi, 2020-07-02:
Thanks for your comment.
Have to try which solution suits best for me.
I wanted to use the cables as they are (connection is safest), so maybe I'll try to mount the connectors first.

hbitter, 2020-07-05:
Hi,
I just came back from a trip a saw what I messed up. I'm really sorry for that.
Please, give me a PM to my mail address helmut.bitter@t-online.de. I will think about how I can compensate it.
Sorry for that.
Regards,
Helmut

hbitter, 2020-07-06:
Hi,
you can try to mount the RJ11/12 connectors as I described in the attached PDF. Please, give me a personal mail to my address above.
Regards,
Helmut
[A possible fix.pdf](https://github.com/hbitter/DNMS/files/4881359/A.possible.fix.pdf)

closed 2020-07-14 by hbitter