hbitter/DNMS #11: Teensy 4.0 Loader DNMS hex-File error line 1

kokoloresBit, 2020-07-14:
Hallo, ich versuche verzweifelt mit dem Teesny Loader 1.52
die verschiedenen Hex-Files DNMS V1.x. x oder Vt.0.4 zu laden.
Der Loader gibt beim Öffnen der Files die Meldung "ihex: parse error line 1" aus.
Was mache ich falsch?
Vielen Dank für die Hilfe im Voraus.
Gruß Thorsten

hbitter, 2020-07-14:
Hallo Thorsten,
sind die herunter geladenen Files O.K.? Am besten geht das, wenn man das gesamte Repository als Zip herunterlädt.
Schau Dir mit einem Hex-Editor z.B. mal DNMS V1.0.5.ino.hex an, sieht das so aus, wie im angehangenen Bild?
[image]

Wenn nicht, der Tipp das gesamte Repository als Zip File herunterladen.

USB-Verbindung zum Teensy ist O.K.?

Viele Grüße
Helmut

hbitter, 2020-07-14:
Hallo Thorsten,
noch ein Tipp. Auf der Teensy Loader Seite für Windows von PJRC (https://www.pjrc.com/teensy/loader_win10.html) gibt es ein Blink Beispiel fertig übersetzt als Hex für alle Teensies. Versuch damit mal Deinen Teeny zu flashen. Dann wissen wir, dass die Verbindung zum Teensy O.K. ist und es prinzipiell funktioniert.
Viele Grüße
Helmut

kokoloresBit, 2020-07-15:
Hallo Helmut,
tatsächlich, das herunterladen als Zip-File ist die Lösung. Ich habe per "Link speichern unter" die hex Files kopiert. Das führt dazu, das der Hex-Datei ein Text hinzugefügt wird: Siehe Auszug..
[image]
Nach diesem Text folgt später der Text zur Firmware.
Vielen Dank für die schnelle Hilfe.
Viele Grüße 
Thorsten

closed 2020-07-16 by hbitter