hbitter/DNMS #12: Kürzere Messintervalle bzw. echte "Livedaten" möglich?

gidhap, 2020-07-16:
Hallo, 
Ich möchte neben der Standardauswertung die Messwerte in kürzeren Abständen erfassen um kurze Lärmspitzen zu identifizieren.
Bisher hatte ich mit meinem Sound Meter eine Sekunde bzw. 0.5 Sekunden, leider konnte das keine Langzeitaufzeichnung.
Wie kann ich echte Livedaten des Mikrofons in 0,5 oder 1 sek angezeigt bekommen bzw abgreifen?

hbitter, 2020-07-16:
Hallo gidhap,

Kurz zur Funktionsweise des DNMS. Es wird intern ca. alle 35ms der aktuelle A-bewertete Schalldruckpegel berechnet aus dem RMS Wert aller digitalisierten Werte dieses Intervalls (44,1 kHz, 16-bit Abtastung). Diese Werte werden weitergerechnet zu dem LAeq Wert bis der Sensor abgefragt wird (z.B. alle 150 Sekunden im Sensor.Community Netz). Alle 35ms wird auch der neue Schalldruckpegel mit dem bisherigen niedrigsten und bisherigen höchsten verglichen und neu abgespeichert, wenn der neue Wert kleiner bzw. größer ist als der bisherige. Diese Werte werden bei Abfrage als LAmin bzw. LAmax ausgegeben d.h. man hat mit LAmax die Spitze in dem abgefragten Intervall. 
Der DNMS ist nicht auf ein Abfrageintervall festgelegt, dies wird vom abfragenden System gesteuert d.h. DNMS kann z.B. im Sekundentakt oder auch jede Stunde abgefragt werden.
Im Sensor.Community Netz ist das Abfrageintervall fest auf 150 Sekunden gesetzt, damit werden die Sensoren "überwacht", ob sie noch aktiv sind und andereseits wird häufigeres senden geblockt, damit es nicht zu überlastungen kommt.
Die Konfiguration der Firmware der NodeMCU erlaubt die freie Wahl des Messintervalls:
[image]

Allerdings in der Standardversion der Firmware nur bis zu 5 Sekunden herunter. Wenn die Firmware anpasst und selbst übersetzt geht natürlich auch 1 Sekunde. 
Übertragungen ins Sensor.Community Netz sind damit natürlich nicht möglich aber man kann andere Möglichkeiten der Datenspeicherung über dei weiteren APIs benutzen. Ich benutze zu Testzwecken eine Influx Datenbank aud einem lokalen PC. Das kann man direkt in der Firmware einrichten:
[image]

Anzeige der Werte z.B. mittels Grafana (InfluxDB und Grafana gibt es in kostenlosen Versionen) sieht dann so aus:
[image]

Da Du ja zwei DNMS betreibst, könntest Du die werte von einen ja lokal im Sekundentakt mittels InfluxDB aufzeichnen.

Bei weiteren Fragen kannst Du Dich auch gerne direkt an mich wenden helmut.bitter@t-online.de
Viele Grüße
Helmut

gidhap, 2020-07-16:
DANKE Helmut, habe Dir noch eine mail geschrieben

closed 2020-07-17 by hbitter