hbitter/DNMS #13: Display SSD1306 und Webseite zeigen keine Messwerte

kokoloresBit, 2020-07-19:
Hallo, 
ich habe zuerst die Version Vt.0.4 ausprobiert. Die Schalldruckpegelwerte werden an den PC übertragen.
Ich habe die Version V1.0.4 geladen. Ich betreibe den Teensy  V4.0 an einem ESP8266 V3.
Ich wollte das Display SSD1306 wie in der angefügten Darstellung betreiben. 
Somit liegt der Teensy mit dem Display parallel am I2C Bus (D3, D4) des ESP8266 . 
Mit angeschlossenem Display werden keine Messwerte auf der Webseite und im Display angezeigt.
Wenn ich das Display abklemme werden die Messwerte auf der Webseite dargestellt.
Vielen Dank im Voraus für die Unterstützung.
Gruss Thorsten

hbitter, 2020-07-19:
Hallo Thorsten,
Du schreibst: "Ich wollte das Display SSD1306 wie in der angefügten Darstellung betreiben." Kannst Du bitte die Darstellung noch senden.

1.Funktioniert das Display ohne DNMS, wird dann Information auf dem Display angezeigt?
2.Wenn das Display angeschlossen ist, könntest Du die NodeMCU über USB an einen PC anschliessen und prüfen, ob dann Debug ausgaben an den PC gehen (Terminalprogramm am PC)? Nach jedem Messintervall werden die Werte auch über die Debug Schnittstelle ausgegeben.
Viele Grüße
Helmut

hbitter, 2020-07-20:
Hallo Thorsten,
noch eine Frage: Welche Firmware Version nenutzt Du auf der NodeMCU? Meine Empfehlung ist, die Beta Version (z.Zt. NRZ-2020-130-B6) zu benutzen.
Viele Grüße
Helmut

kokoloresBit, 2020-07-20:
> Hallo Thorsten,
> Du schreibst: "Ich wollte das Display SSD1306 wie in der angefügten Darstellung betreiben." Kannst Du bitte die Darstellung noch senden.
> 
> 1.Funktioniert das Display ohne DNMS, wird dann Information auf dem Display angezeigt?
> 2.Wenn das Display angeschlossen ist, könntest Du die NodeMCU über USB an einen PC anschliessen und prüfen, ob dann Debug-Ausgaben an den PC gesendet werden (Terminalprogramm am PC)? Nach jedem Messintervall werden die Werte auch über die Debug Schnittstelle ausgegeben.
> Viele Grüße
> Helmut

Hallo Helmut,
Sorry, die Darstellung hatte ich vergessen einzufügen.
[image]

zu 1. Ja, mit und ohne DNMS funktioniert das Display, nur die Messwerte für den Schalldruck werden nicht angezeigt. 
[image]

zu 2. Das probiere ich jetzt gleich aus.
Aber ich habe auf der Webseite folgende Anzeige:
[image]

Viele Grüße
Thorsten

kokoloresBit, 2020-07-20:
> Hallo Thorsten,
> noch eine Frage: Welche Firmware Version benutzt Du auf der NodeMCU? Meine Empfehlung ist, die Beta Version (z.Zt. NRZ-2020-130-B6) zu benutzen.
> Viele Grüße
> Helmut

Hallo Helmut,
ja, ich nutze die Version Firmware: NRZ-2020-130-B6/DE (May 8 2020 11:24:33).
Viele Grüße 
Thorsten

kokoloresBit, 2020-07-20:
> Hallo Thorsten,
> Du schreibst: "Ich wollte das Display SSD1306 wie in der angefügten Darstellung betreiben." Kannst Du bitte die Darstellung noch senden.
>
> 1.Funktioniert das Display ohne DNMS, wird [...]

Hallo Helmut,
zu 2. Das Terminal Programm gibt aus " output values to display..."
Viele Grüße 
Thorsten

kokoloresBit, 2020-07-20:
Hallo Helmut,
ich habe jetzt an den I2C Bus den Sensor BME280 parallel zum Display SSD1306 und dem
DNMS angeschlossen. Das Display zeigt die Temp., den Luftdruck und die Luftfeuchte an.
Und über die USB-Verbindung werden die Daten Temp., den Luftdruck und die Luftfeuchte 
auch im Terminal-Programm am PC angezeigt. 
Vom DNMS Sensor werden keine Daten am Display oder im Terminal Program angezeigt.

Dann habe ich das Display abgeklemmt. Jetzt werden nur die Daten vom Sensor BME280 
an das Terminal Programm übertragen.
Im Terminal Programm steht " Trying DNMS sensor on 0x55H not found"

Dann habe ich den Sensor BME280 abgeklemmt. Jetzt zeigt das Terminal  Programm
"Trying DNMS sensor on 0x55H ... found: DNMS Version 1.0.4"
Die Daten zum Schalldruck werden an das Terminal Programm übertragen.
[image]
[image]

Könnte es ein Problem mit dem Timing auf dem I2C Bus geben?

Vielen Grüße
Thorsten

hbitter, 2020-07-20:
Hallo Thorsten,
vielen Dank für Deine Info. Ich habe vorhin versucht mal wieder ein Display anzuschließen und zu aktivieren leider mit dem gleichen Effekt. Kein paralleler Betrieb vom Display und DNMS. Da werde ich mal in Ruhe mit dem Logic Analyzer schauen müssen. Kann leider einige Tage daueren. 
Viele Grüße
Helmut

hbitter, 2020-07-21:
Hallo Thorsten,
kleiner Zwischenstand: Ich hatte einige Probleme mal wieder eine Konfiguration mit Display zum Laufen zu bringen. Immer wenn ich in der NRZ-2020-130-B6 Firmware das Display konfigurieren wollte, ist die NodeMCU gecrascht, danach natürlich automatisch wieder gestartet (ein Watchdog überwacht das). Zwei Dinge habe ich gefunden:
1. Ich konnte das Display nur in der NRZ-2020-129 Version konfigurieren und dann auf NRZ-2020-130-B6 updaten
2. Es gab dann aber immer noch keine Ausgabe auf dem Display, da zumindest für mein OLD SSD1306 die I²C Adresse in der Firmware nicht richtig ist. Ich musste die von 0x3c auf 0x3d ändern, dann geht es:

[image]
[image]
[image]

Diese Änderung kann aber Display abhängig sein und bei Dir muss es mit der I²C Adresse nicht so sein aber Du kannst es wenn Du willst ausprobieren. Schicke mir eine PM an helmut.bitter@t-online.de und dann kann ich die eine geänderte Version NRZ-2020-130-B6+ schicken zum Flashen.
Bitte vorher die Konfiguration mit Display, BME280 und DNMS eintragen und dann die neue Version flashen.
Viele Grüße
Helmut

closed 2020-09-14 by hbitter