hbitter/DNMS #14: Teensy mit Arduino flashen, Bibliotheken überschneiden sich

Melnilie, 2020-07-23:
Hallo,

ich hab versucht meinen Teensy4.0 mit dem TeensyLoader zu flashen, aber aus irgendeinem Grund ging das nicht.
Also habe ich es mit der Arduino IDE probiert. Das Testprogramm ging auch, aber wenn ich versuche DNMS_V1.0.4 draufzuladen, bekomme ich diesen Fehler

> /tmp/arduino_build_726161/libraries/Wire/WireIMXRT.cpp.o:(.data.Wire2+0x0): multiple definition of `Wire2'
/tmp/arduino_build_726161/libraries/teensy4_i2c-master/i2c_driver_wire.cpp.o:(.bss.Wire2+0x0): first defined here
/home/user/Downloads/arduino-1.8.13/hardware/tools/arm/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: Disabling relaxation: it will not work with multiple definitions
/home/user/Downloads/arduino-1.8.13/hardware/tools/arm/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: Warning: size of symbol `Wire2' changed from 116 in /tmp/arduino_build_726161/libraries/teensy4_i2c-master/i2c_driver_wire.cpp.o to 112 in /tmp/arduino_build_726161/libraries/Wire/WireIMXRT.cpp.o
/tmp/arduino_build_726161/libraries/Wire/WireIMXRT.cpp.o: In function `Print::availableForWrite()':
/home/user/Downloads/arduino-1.8.13/hardware/teensy/avr/libraries/Wire/WireIMXRT.cpp:235: multiple definition of `Wire1'
/tmp/arduino_build_726161/libraries/teensy4_i2c-master/i2c_driver_wire.cpp.o:/home/user/Arduino/libraries/teensy4_i2c-master/src/i2c_driver_wire.cpp:123: first defined here
/home/user/Downloads/arduino-1.8.13/hardware/tools/arm/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: Warning: size of symbol `Wire1' changed from 116 in /tmp/arduino_build_726161/libraries/teensy4_i2c-master/i2c_driver_wire.cpp.o to 112 in /tmp/arduino_build_726161/libraries/Wire/WireIMXRT.cpp.o
/tmp/arduino_build_726161/libraries/Wire/WireIMXRT.cpp.o: In function `Print::availableForWrite()':
/home/user/Downloads/arduino-1.8.13/hardware/teensy/avr/libraries/Wire/WireIMXRT.cpp:235: multiple definition of `Wire'
/tmp/arduino_build_726161/libraries/teensy4_i2c-master/i2c_driver_wire.cpp.o:/home/user/Arduino/libraries/teensy4_i2c-master/src/i2c_driver_wire.cpp:123: first defined here
/home/user/Downloads/arduino-1.8.13/hardware/tools/arm/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld: Warning: size of symbol `Wire' changed from 116 in /tmp/arduino_build_726161/libraries/teensy4_i2c-master/i2c_driver_wire.cpp.o to 112 in /tmp/arduino_build_726161/libraries/Wire/WireIMXRT.cpp.o
collect2: error: ld returned 1 exit status
Fehler beim Kompilieren für das Board Teensy 4.0.

Ich wollte jetzt auch nichts an den Librarys ändern, deshalb habe ich da noch nichts versucht. 
Wäre cool, wenn mir jemand helfen könnte.

hbitter, 2020-07-23:
Hallo Melnilie,
1. Problem Teensy4.0 mit TeensyLoader flashen
Sind die herunter geladenen Files O.K.? Am besten geht das, wenn man das gesamte Repository als Zip herunterlädt.
Schau Dir mit einem Hex-Editor z.B. mal DNMS V1.0.5.ino.hex oder DNMS V1.0.4.ino.hex an, sieht das so aus, wie im angehangenen Bild?
[image]

Wenn nicht, der Tipp das gesamte Repository als Zip File herunterladen und es dann das Flashen mit TeensyLoader probieren.

2. Problem Firmware in der Arduino IDE komplieren
Da sind einige kleine Anpasungen notwendig:

1. You have to install Arduino IDE (Version 1.8.13) an Teensyduino (1.53)

For Teensy3.6:

2. As you can see in the code the lib i2c_t3.h is included, this lib is used instead of of the Wire.h lib. You can download that lib from github as a zip and install it in Arduino.

3. In the Teensyduino Audio lib you have to change all "#include Wire.h" to "#include <i2c_t3.h>". Where the Teensyduino Audio lib is located depends on your used Operating Sytem and your installation. On my Linux it located under: /home/helmut/Arduino-SW/arduino-1.8.13/hardware/teensy/avr/libraries/Audio/. In that directory you will find several files named: control_xxxxx.cpp. In all those files change "#include Wire.h" to "#include <i2c_t3.h>"

It should now compile.

 

For Teensy4.0:

2. As you can see in the code instead of the Wire.h lib another I²Cl ib is used for Teensy4.0. It is the lib teensy4_i2c. You can download that lib from github as a zip and install it in Arduino.

3. Now you have to change in the Teensyduino Audio lib all "#include Wire.h" to #include <i2c_driver.h> and #include <i2c_driver_wire.h> as described under 3. fpr Teensy3.0

It should now compile.

Ich hoffe, damit geht es. Bitte Rückmeldung an mich auch gerne direkt an helmut.bitter (at) t-online.de

Viele Grüße
Helmut

Melnilie, 2020-07-23:
Hallo Helmut, 

Danke für die Übersicht.
Ich hatte mir nachdem ich hier gepostet habe, den TeensyLoader nochmal neu installiert und auch die Dateien neu runtergeladen und jetzt ging das Flashen auch. 
Ich bekomme jetzt über die serielle Schnittelle die Mikrofondaten, aber wenn ich das ganze an den NodeMCU anschließe und DNMS auf der Konfigurationsseite anklicke, bekomme ich über die Webseite keine Daten.
Ich hab auch schon an SCL und dem SDA Anschluss einen 5k Widerstand an 3,3V angeschlossen und das hat auch nicht geholfen . Hast du da vielleicht noch eine Idee?

Viele Grüße
Mel

hbitter, 2020-07-23:
Hallo Mel,

was heißt "Ich bekomme jetzt über die serielle Schnittelle die Mikrofondaten"? Welche Firmware ist da auf dem Teensy4.0? Und welche serielle Schnittstelle ist gemeint, die vom Teensy4.0 oder von der NodeMCU?
Wenn Du die DNMS_V1.0.4 oder .5 Firmware auf dem Teensy hast, werden über die serielle Schnittstelle keine Werte ausgegeben. Das Testprogramm DNMS_Vt.0.4 gibt die Werte über die serielle Schnittstelle aus aber hat keine Kommunikation zur NodeMCU (dient zum Test ohne NodeMCU). Wenn es so ist, bitte die DNMS_V1.0.4 oder .5 Firmware auf dem Teensy flashen.
Wenn NodeMCU und Teensy geflasht sind, serielle Schnittstelle der NodeMCU anschließen und mit Terminalprogramm den Debug Output ansehen. Evtl. einmal Reset Taste auf der NodeMCU drücken um den kompletten Vorgang beim Start zu betrachten.
Output sollte dann ungefähr so aussehen:
[image]

Wichtig sind die Zeilen, dass sich die NodeMCU mit dem WLAN verbindet und dann, wenn DNMS konfiguriert ist (Read DNMS ...) die nächste Zeile, die zeigt, ob der DNMS gefunden wurde.

Kommst Du soweit?
Viele Grüße
Helmut

Melnilie, 2020-07-27:
Hallo Helmut,

Entschuldige die späte Antwort, ich komm wegen Cornoa nicht so oft in unser Lab.
Ich habe die Hex Dateien angesehen und die sehen richtig aus, warum das nicht über den Teensyloader geht, kann ich mir nicht erklären. 
Allerdings habe ich jetzt die Audio Bibliotheken geändert und dann die Firmware auf den Teensy geladen.
Im Output steht leider immer noch, dass er den DMNS nicht findet, aber der Feinstaubsensor und der Temperatursensor gehen.

Viele Grüße
Mel

hbitter, 2020-07-27:
Hallo Mel,
 
ich bin z.Zt. unterwegs und melde mich Anfang nächster Woche wieder.
 
Viele Grüße
Helmut

überschneiden sich (#14)
Datum: 2020-07-27T09:11:14+0200
 
 
 
 

<https://github.com/hbitter/DNMS/issues/14#issuecomment-664163582> , or 
unsubscribe
"https://github.com/hbitter/DNMS/issues/14#issuecomment-664163582", "url": 
"https://github.com/hbitter/DNMS/issues/14#issuecomment-664163582", "name": 
"View Issue" }, "description": "View this Issue on GitHub", "publisher": { 
"@type": "Organization", "name": "GitHub", "url": "https://github.com" } } 
]

imaluschi, 2020-07-29:
Selbes Problem mit Teensy 3.6.
Zusätzlich zu den Änderungen an der Teensyduino Audio lib musste ich die Datei WireKinetis.h ändern:
Arduino-Installation.../arduino-1.8.13/hardware/teensy/avr/libraries/Wire/WireKinetis.h
Auskommentieren der #define Wire_... bei Teensy 3.6. Das sind die Definitionen, die in der i2c_t3-Library schon gemacht wurden.
```
// Teensy 3.6
#elif defined(__MK66FX1M0__)
/*#define WIRE_IMPLEMENT_WIRE
#define WIRE_IMPLEMENT_WIRE1
#define WIRE_IMPLEMENT_WIRE2
//Wire3 is seldom used on Teensy 3.6
//#define WIRE_IMPLEMENT_WIRE3
#define WIRE_HAS_START_INTERRUPT
#define WIRE_HAS_STOP_INTERRUPT
*/
#endif
```
Soeben geflasht ;-)

Melnilie, 2020-07-30:
Hallo,

Ich benutze den Teensy 4.0 hab das jetzt in der WireKinetis Bibliothek trotzdem mal auskommentiert, leider keine Veränderung.

imaluschi, 2020-07-30:
Hallo,

bei mir funktioniert es auch mit Teensy 4.0 nach folgenden Schritten:
Wie von Helumt erklärt, die Schritte für Teensy 4.0:
https://github.com/hbitter/DNMS/issues/14#issuecomment-662954198
Die von mir beschriebene Änderung für Teensy 3.6 in der WireKinetis.h wurde so belassen, für Teensy 4.0 scheint dies irrellevant zu sein, da ja
```
// Teensy 3.6
#elif defined(__MK66FX1M0__)
```
Da gibts kein `if defined` für Teensy 4.0.

Ich hab die Audio Lib (in Arduino-Installation.../arduino-1.8.13/hardware/teensy/avr/libraries/Audio), allle control*.cpp nun gleich die Zeile `#include <Wire.h>` so abgeändert:
```
//#include <Wire.h>
//#include <i2c_t3.h> //Teensy 3.6
#include <i2c_driver.h> //Teensy 4.0
#include <i2c_driver_wire.h>
```
Theoretisch müsste es möglich sein, auch dort überall eine if-Abfrage machen (Teensy 3.6 oder Teensy 4.0).

hbitter, 2020-07-30:
Also ich empfehle fogendes Vorgehen beim Teensy4.0
1. Testprogramm DNMS_Vt.0.4 auf Teensy flashen und Output über USB/Terminalprogramm ansehen
2. Flashen des Teensy mit DNMS_V1.0.4.ino.hex. Es gibt eigentlich keinen Grund, dass das Flashen nicht geht, wenn es mit dem Testprogramm ging. Wenn man das DNMS_V1.0.5.ino.hex nimmt, wird über USB dasAudio Signal ausgegeben, das man z.B. mit Audacity ansehen/aufzeichnen. Damit ist dann sichergestellt, dass Das Programm auf dem Teensy läuft. (das geht natürlich auch mit dem Teensy3.6 - DNMS_V1.1.1.ino.hex).
3. Zum Test der I²C-Kommunikation gibt es für die NodeMCU einen I²C Scanner:

`/* 
* i2c_port_address_scanner
* Scans ports D0 to D7 on an ESP8266 and searches for I2C device. based on the original code
* available on Arduino.cc and later improved by user Krodal and Nick Gammon (www.gammon.com.au/forum/?id=10896)
* D8 throws exceptions thus it has been left out
*
*/

#include <Wire.h>

void setup() {
  Serial.begin(9600);
  while (!Serial);             // Leonardo: wait for serial monitor
}

uint8_t portArray[] = {16, 5, 4, 0, 2, 14, 12, 13};
//String portMap[] = {"D0", "D1", "D2", "D3", "D4", "D5", "D6", "D7"}; //for Wemos
String portMap[] = {"GPIO16", "GPIO5", "GPIO4", "GPIO0", "GPIO2", "GPIO14", "GPIO12", "GPIO13"};

void scanPorts() { 
  for (uint8_t i = 0; i < sizeof(portArray); i++) {
    for (uint8_t j = 0; j < sizeof(portArray); j++) {
      if (i != j){
        Serial.print("Scanning (SDA : SCL) - " + portMap[i] + " : " + portMap[j] + " - ");
        Wire.begin(portArray[i], portArray[j]);
        check_if_exist_I2C();
        wdt_reset(); // nodemcu is alive
      }
    }
  }
}

void check_if_exist_I2C() {
  byte error, address;
  int nDevices;
  nDevices = 0;
  for (address = 1; address < 127; address++ )  {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.
    Wire.beginTransmission(address);
    error = Wire.endTransmission();

    if (error == 0){
      Serial.print("I2C device found at address 0x");
      if (address < 16)
        Serial.print("0");
      Serial.print(address, HEX);
      Serial.println("  !");

      nDevices++;
    } else if (error == 4) {
      Serial.print("Unknow error at address 0x");
      if (address < 16)
        Serial.print("0");
      Serial.println(address, HEX);
    }
  } //for loop
  if (nDevices == 0)
    Serial.println("No I2C devices found");
  else
    Serial.println("**********************************\n");
    delay(5000);           // wait 1 seconds for next scan, did not find it necessary
}

void loop() {
  Serial.println("\n\nI2C Scanner to scan for devices on each port pair D0 to D7");
  scanPorts();

  delay(5000);

}
`
Den I²C Scanner auf die NodeMCU laden und laufen lassen. DNMS hat die I²C Adtresse 0x55 und sollte dann gefunden werden. Wenn nicht, gibt es ein Problem mit der I²C Verbindung (GND, SCL, SDA). 
Stromversorgung Teensy? Das ist am Blinken des Teensy Boards zu sehen.

Ich hoffe, Ihr kommt damit weiter.
Viele Grüße
Helmut

Melnilie, 2020-08-06:
Hallo Helmut,

Ich habe das mit einem I2C Scanner ausprobiert und er findet die Verbindung wohl nur manchmal. Es liegt also wohl an meinem Board . Ich hab mir jetzt mal ein neues bestellt und schau, ob es damit besser funktioniert.
Danke für deine Hilfe.

Viele Grüße

closed 2020-09-07 by hbitter