hbitter/DNMS #15: Idee: Fahrzeugzählung mit DNMS?

linuxlurak, 2020-09-04:
Hallo

Ich werde nächstens die ersten DNMS bauen können. Mir ist folgende Idee gekommen: Könnte man mit dieser Technologie auch Fahrzeugzählungen vornehmen? Wäre es sodann denkbar, dass man einzelne Fahrzeuge einer Kategorie zuordnen kann? Ich denke da typischerweise an Töffs, die durch ihr Knattern eventuell identifizierbar wären.

Danke für Eure Gedanken zu dieser Idee.
Gruss

Franz4596, 2020-09-06:
hi Linuxlurak,
aktuell berechnet der teensy-Controller "nur" den A-bewerteten Schallpegel LAeq, LAmin, LAmax. Eine Pegelüberschreitung kann man natürlich feststellen und zählen, vgl. https://openmaps.online/noise/ .
Eine Kategoriezuordnung könnte über Spektrenvergleiche oder Korrelationsverfahren erfolgen. Eine Frequenzanalyse (FFT) soll vielleicht implementiert werden. 
mfg franz

etho50, 2020-09-07:
Das Potential für solche Zusatzfunktionalitäten ist vorhanden. Wir haben nicht ohne Grund den Teensy-Prozessor als speziellen sehr leistungsstarken 'Lärmprozessor' (u.a. mit Gleitpunktarithmetik in Hardware) ausgewählt und die Funktionen Lärmauswertung von der Kommunikation über den ESP getrennt. Das Potential müsste nur gehoben werden. Im Klartext: Wir bräuchten dafür Software-Entwickler am besten mit Erfahrung im Bereich der Mustererkennung mit Methoden der KI. Das Thema Spektren als Basis dafür wollen wir aber ohnehin angehen. Ob man alleine mit Korellationsrechnungen der Spektren  schon befriedigende Ergebnisse beim Straßenverkehr bekommen kann, müsste man testen. Ich bin da etwas skeptisch. Bei Bahnlärm ist die Erkennung von Zügen v.a. nachts und wenn der Sensor nahe am Gleis angebracht ist, näherungsweise durchaus schon jetzt möglich. Sie sind meist als LAeq-Peaks gut erkennbar. Wenn dann noch  LAmax hoch geht und möglicherweise z.B. bei langen Güterzügen auch LAmin reagiert, kann man das Ergebnis zusätzlich absichern.  Es gibt auch Sensoren im Netz, bei denen man dies gut sehen kann. Z.B. der Sensor 44836. Da kann man sogar S-Bahnen und Güterzüge voneinender unterscheiden. Der S-Bahn-Fahrplan ist gut erkennbar (kleinere Peaks), die größeren Peaks sind Güterzüge.  Man könnte sogar die Fahrtrichtung der Züge ermitteln, wenn man mehrere Sensoren auswertet, die in größerem Abstand an der jeweiligen Strecke angebracht sind. Leider ist der Sensor gerade Offline, weil er repariert bzw. ausgetauscht werden muss. Aber an der Nachbarstation 36217 ist dies auch erkennbar. Bei Flugzeugen könnte ich mir das auch vorstellen, aber auch nur, wenn das Geräusch an dem Standort sehr signifikant ist.  
Leider ist es schwer, Leute zu gewinnen, die bei unserem Projekt bei der Weiterentwicklung gerade solcher Apps mitmachen wollen und dann auch eine längere Zeit dabei bleiben. Das ist Schade. Wir sind für jeden dankbar, der uns hilft. Tatsächlich ist es auf Dauer kaum möglich, das bisherige Niveau zu halten, wenn wir unser Team nicht erweitern können. 
Grüße
Ewald

linuxlurak, 2020-09-13:
Vielen Dank für Eure eingehenden Rückmeldungen. Ich merke, dass ich da an meine Grenzen stosse. Ich bin kein Ingenieur, aber wenigstens passionierter Device-Hacker und kann zur Not auch SMD Löten und ein PCB designen. Ich könnte für Firmwaretest oder das Bearbeiten von Anleitungen (zB DE/EN) etwas Zeit aufbringen, _guten_ Code schreiben oder mich mit den theoretischen Grundlagen der Anwendungen auseinandersetzen übersteigt meine Kenntnisse aber vermutlich.

atze09, 2020-09-21:
> hi Linuxlurak,
> aktuell berechnet der teensy-Controller "nur" den A-bewerteten Schallpegel LAeq, LAmin, LAmax. Eine Pegelüberschreitung kann man natürlich feststellen und zählen, vgl. [...]

Hallo Franz,
Lässt sich diese Datenbank (https://openmaps.online/noise/) auch über einen längeren Zeitraum visualisieren (meine InfluxDB ist leider gerade offline).
Das wäre super.
Freundliche Grüße,
Dominik

atze09, 2020-09-21:
> Das Potential für solche Zusatzfunktionalitäten ist vorhanden. Wir haben nicht ohne Grund den Teensy-Prozessor als speziellen sehr leistungsstarken 'Lärmprozessor' (u.a. mit Gleitpunktarithmetik in [...]

Hallo Ewald,

Vielen Dank für die zahlreichen Informationen.
Ich bin leider Laie wenn es um Programmierung geht. Bezüglich FFT hätte ich zumindest die theoretischen Voraussetzungen falls das etwas hilft, könnte ich hier ab nächsten Jahr unterstützen.

Freundliche Grüße,
Dominik

Franz4596, 2020-09-22:
> Hallo Franz,
> Lässt sich diese Datenbank (https://openmaps.online/noise/) auch über einen längeren Zeitraum visualisieren (meine InfluxDB ist leider gerade offline).
> Das wäre super.
> Freundliche [...]

Hi Dominik,
Die map wurde von CodeforCologne erstellt, Infos dazu siehe "?" auf http://openmaps.online/noise/, 
bzw. code https://github.com/codeforcologne/OpenNoiseMap. 
Du kannst aber auf alle Lärmdaten von Sensor.Community zugreifen (madavi.de speichert akt. kein Lärm).
oder Grafik in https://laerm.citysensor.de/
franz

atze09, 2020-09-23:
> Hi Dominik,
> Die map wurde von CodeforCologne erstellt, Infos dazu siehe "?" auf http://openmaps.online/noise/,
> bzw. code https://github.com/codeforcologne/OpenNoiseMap.
> Du kannst aber auf alle [...]

Thank you!

closed 2020-10-30 by hbitter