hbitter/DNMS #16: Wo PCBs bestellen?

ituri, 2020-09-04:
Hallo zusammen,

verfolge das Lärmsensor-Projekt nun schon einige Monate und möchte mich endlich auch an die eigene Umsetzung machen. Über den Bezug des Mikrofons findet man ja hier auf GitHub schon einige wertvolle Hinweise. Für den Bezug des PCBs konnte ich aber weder auf luftdaten.info noch hier etwas finden. Könnt Ihr mir Erfahrungswerte mit Dienstleistern nennen, wo ich das PCB bestellen kann? Hab gerade mal ganz naiv den ersten Google-Treffer versucht (https://de.beta-layout.com/), der aber für das T4-nodemcu Board 77,54EUR+Versandkosten verlangt. 
Geht es auch günstiger, gerne auch in Asien? Könnt ihr irgendwelche Anbeiter empfehlen?

Viele Grüße!

hbitter, 2020-09-04:
Hallo ituri,
das geht günstiger in China z.B. bei https://jlcpcb.com/.
Ich habe gerade noch einige Boards zur Verfügung, wenn Du 1 -2 benötigst, schicke mir eine Mail (helmut.bitter (at) t-online.de) was Du benötigst.
Viele Grüße
Helmut

ituri, 2020-09-04:
Danke Helmut, ich melde mich!

Franz4596, 2020-10-04:
Ggf. bei mir für Sammelbestellung melden (Platine ~1€/Stk)  franz.hoefle(a)online.de

Franz4596, 2020-10-31:
Gebe einzelne Platinen für 0,60€/Stk+Porto u. ggf. Mikro an Bastler ab <franz.hoefle(a)online.de>