hbitter/DNMS #17: SDS011 anstatt SPS30 möglich?

linuxlurak, 2020-09-04:
Kann ich den SDS011 anstelle eines SPS30 verwenden? Auch mit DNMS-Board? 
Gruss

hbitter, 2020-09-04:
Ja, natürlich. Der SDS011 (ebenso der SPS30) werden von der NodeMCU abgefragt und nicht vom Teensy Prozessor des DNMS.
Auf den Airrohr Boards sind beide Buchsen einmal für den SPS30 und einmal für den SDS011. An den Kombi Boards (Teensy3.6+NodeMCU bzw. Teensy4.0+NodeMCU) ist nur eine Buchse für den SPS30. Ein SDS011 kann aber mit etwas Lötaufwand an der NodeMCU angeschlossen werden (Anschluss wie auf der Sensor.community Seite gezeigt). Bei einer solchen Konfiguration mit dem SDS011 ist es aber unbedingt wichtig für die notwendige akustische Entkopplung zwischen dem Mikro vom DNMS und dem SDS011 zu sorgen. Der Ventilator dea SDS011 ist ja so schon deutlich zu höhren und das bekommt das Mikro ja auch mit. Aus diesem Grund haben wir eigentlich auf die Trennug zwischen Airrohr und DNMS Board mit einer auch längeren Kabelverbindung um das DNMS entkoppelt und akustisch gut zu positionieren.

linuxlurak, 2020-09-04:
Danke für Deine Antwort. 

Zu Deinem Input betr- Lüftergeräusche: Ich denke im Moment ein Setup an, das mit 75er HT-Rohr und Boden arbeitet. Hört das Mikro also den SDS011 im HT Bogen hinte? Hätte für die Ansaugung direkt im HT-Bogen ein zusätzliches Loch gemacht. Bringt allenfalls zusätzliche Isolation um den SDS011 etwas? Oder ist das Problem, dass der SDS011 den Plastik berührt und so seine Lüftergeräusche aufs Mikro überträgt?

Zum DNMS-Board T.4+NodeMCU (habe bei dirtypcb einen Batch davon geordert): Ist es möglich, den Anschluss für den SP30 zu benutzen und in der Firmware die Pins einzustellen? Die Firmware, die ich vom Feinstabsensor her kenne, ist einfach gehalten (was natürlich grosse Vorteile hat für eine breite Community). Dort hätte ich keine Einstellungsmöglichkeiten gesehen. Sonst also wie von Dir erwähnt an die entsprechenden Pins löten. Wär auch kein Problem; Steckplätze benutzen wär natürlich sehr praktisch.
Schönen Abend!
Gruss

hbitter, 2020-09-04:
Also ich sehe zwei mögliche Problempunkte mit dem DN75 Rohr - ohne dass ich die Anordnung genau verstehe.

- Schall/Vibrationsübertragungen des SDS011 an das Mikro - ja diese Möglichkeit besteht wenn beide (SDS011 und Mikro fest am gleichen Kunststoffkörper angekoppelt sind

- wenn das DNMS Mikro mit dem 1/2" Rohr direkt in den Abschluss des DN75 Rohrs mündet, entstehen starke Reflexionen von der Abschlussplatte des DN75 Rohrs zum Mikro hin, die dei Messwerte verfälschen. Dies ist auch stark Frequenzabhängig. Bei Messungen haben wir Abweichungen vonbis zu 5 dBA gemessem. Empfehlung: Das Abschlussrohr zum 1/2" Mikro Rohr möglichst klein halten (nicht mehr wie DN50).

Die beste akustische Entkopplung ergibt sich mit einem abgesetzten DNMS.

Beim Kombi Board kann der Anschluss des SPS30 nicht (in keinem Fall) für den Anschluss eines SDS011 benutzt werden. Das SPS30 wird am I²C Bus angeschlossen, wie auch das DNMS. Das SDS011 benutzt eine separate serielle Schnittstelle UART mit TX und RX Leitung), auf der keine anderen Geräte angeschlossen werden können. SDS011 geht nur, wenn er an die entsprechenden Pins an der NodeMCU angeschlossen (angelötet) wird.

Wenn Du weitere Fragen hast, auch gerne per PM an helmt.bitte (at) t-online.de

Viele Grüße
Helmut

closed 2020-09-23 by hbitter