hbitter/DNMS #21: Grundsätze für die Standortwahl des DNMS-Mikrofons

linuxlurak, 2020-09-20:
Hallo

Ich wäre froh, wenn mir jemand die wesentlichsten Grundsätze zur Standortwahl angeben könnte. Ich würde den Sensor am liebsten an der Decke auf dem Balkon montieren. Die Decke ist mit Spaghettiplatten (Holzwolle-Leichtbauplatte) ausgekleidet.

Vielen Dank und Gruss

etho50, 2020-09-21:
Hallo linuxlurak,

eine solche Beschreibung gibt es seit einiger Zeit in englisch auf der sensor.community-Seite mit Bildern:

https://sensor.community/en/sensors/dnms/
(ganz hinten im Dokument)

Die akustische Beurteilung eines Anbringungsorts ist leider nicht ganz einfach. Der wichtigste Grundsatz ist, das Mikro möglichst frei anzubringen, möglichst weit weg von schallharten Flächen, die reflektieren könnten. 

Die Anbringung an der Decke eines Balkons könnte dadurch etwas problematisch sein. Auf jeden Fall sollte das Mikro 'ins Freie' hinausragen, je weiter ums so besser, wobei mir klar ist, dass dies in unseren Wohngebieten gewisse Grenzen hat. Ggf kann man Reflexionen der Decke auch mit Schaumstoff etwas abmildern.  Es hängt auch von der Form des Balkons ab. Je mehr er Loggia-Charakter hat - also an allen Seiten Wände hat, wird es problematisch, Das kann dann ein Verstärker sein.  Man kann auch etwas experimentieren und die Wirkung bei unterschiedlichen Positionen testen. Dann kann man zumindest grobe Fehlmessungen vermeiden. 

Im Übrigen sei an dieser Stelle erwähnt, dass es auch die Möglichkeit gibt, in der Konfiguration des DNMS eine Korrektur  anzubringen. Dafür gibt ein zusätzliches Feld im Konfigurationsmenu. Das ist allerdings eher etwas für eine eventuelle Drift des Sensors nach längerer Zeit und wenn man sehr genau messen will. Nach einem Jahr kontinuierlicher Messung kann dies in leichter Form vorkommen. 

Grüße
Ewald

rin67630, 2022-05-16:
Übrigens: Unbedingt Vogelschutz beachten !
Eine verliebte Taube, oder eine protzende Amsel, die 2m vorm Mikrofon sitzen können locker den Strßenlärm um 10dB übertönen.
Just my 2 Cents...

rin67630, 2022-05-16:
[Reflexion über Reflexionen.pdf](https://github.com/hbitter/DNMS/files/8699415/Reflexion.uber.Reflexionen.pdf)

Beim MEMS konzept würde sich auch eine Variante für Balkondeckenmontage anbieten. Eine gute möglichkeit Reflexionen zu vermeiden, besteht daraus das Mikrofon **direkt an der reflektierende Fläche** zu montieren. 

Toningenieure die z.B. Orgelmusik in eine Kirche aufnehmen, tun das mit Wandmikrofone. Eine Konferenz am runden Tisch wird besser mit Flachmikrofone aufgenommen.