hbitter/DNMS #22: Empfehlung für den Mikrofon-Windschutz

linuxlurak, 2020-09-20:
Hallo

Welcher Windschutz für das Mikro eignet sich am besten? Genügen simple Schaumstoffschütze oder gar Putzschwämme aus der Küche? Ist ein Fell-Windschutz sinnvoll?

Vielen Dank für Eure Inputs & Gruss

ituri, 2020-09-20:
Ich habe inzwischen [diesen](https://www.amazon.de/gp/product/B07D78H8NB/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1) Satz Windschutz besorgt. Passt gut vom Durchmesser und man kann ihn weit genug nach vorne schieben, damit er nicht direkt auf dem Mikrofon-PCB anliegt.

Franz4596, 2020-09-29:
Bei Aliexpress / ebay gibts Windschutz in verschiedenen Gößen. Den 28x24x10mm (@ituri) finde ich etwas klein. Ich verwende Microphone Foam Sponge Windscreen Mic Cover 45x25x11mm, 
habe auch 55x30x12 hält aber nicht so gut weil 12mm Lochdurchmesser etwas groß - um nacktes Rohr Streifen Kletttape kleben hilft. Habe früher auch schon mal ein Schut aus einem Spülschwamm geschnitten, funktionierte auch -  Dämpfung war gering.
Allerdings habe bei mir schon die Vögel Futter oder Baumaterial aus Windschutz rausgepickt, s.: 
[image]
franz

buv, 2021-01-21:
Die Firma GFS hat den Pickschutz für ihre Mikrofone meiner Meinung nach recht elegant gelöst, indem sie Drahtenden abstehen lässt, die Vögel abhalten sollen: 

https://www.gfs-hofheim.de/pdf/Schallpegel-Monitor_SPM_TWO.pdf

pe0ter, 2021-01-21:
I have used a strip of birdspikes I found in our local hardware store (hornbach)
 [image]

hbitter, 2021-01-21:
Equipment from Gfs looks really nice - it is in a different price range.

Regards,
Helmut

rin67630, 2022-05-16:
Schaut biite auf Issue 47, ich habe eine unkonventionelle Lösung anzubieten, die ein super Windschutz bietet (im Windkanal der TU Düsseldorf getestet) und nicht verrottet...