hbitter/DNMS #25: Alte LAN-Kabel zur Mikrofonverdrahtung: Spricht etwas dagegen?

linuxlurak, 2020-09-27:
Spricht etwas dagegen, alte LAN-Kabel zur Verdrahtung des Mikros zu nehmen? Sie dürften einiges unter den empfohlenen AWG26 liegen (obwohl ich gerade auf [Wikipedia](https://en.wikipedia.org/wiki/Category_5_cable) sehe, dass LAN-Kabel AWG24 sein sollen...). Praktisch daran ist aber, dass so ein Kabel genug Adern hat und enorm verfügbar ist.

hbitter, 2020-09-27:
Na ja, das ganze LAN-Kabel ist schon recht unflexibel oder willst Du die einzelnen Adern aus dem LAN-Kabel nehmen?
Meine Empfehlung für Silikon Kabel kommt vom Verhalten beim Löten, da schrumpft die normale Plastik Ummantelung oft stark zusammen und die Ummantelung in der Nähe der Lötstelle wird recht brüchig. Der Unterschied zwischen AWG26 und AWG24 beträgt ungefähr 0,1mm im Durchmesser (ohne Isolierung), ist also nicht groß.

linuxlurak, 2020-09-27:
Ja, habe ein Kabel mal aufgedröselt und dann von den 8 adern 6 verlötet. Die Isolierung schmilzt recht schnell weg, also rasch arbeiten ist angesagt. Das LAN-Kabel ist nicht sehr flexibel, da hast Du recht. Zum eingiessen sollte das kaum ein Problem sein. Kommt dann danach drauf an, was mit dem Kabel geschieht. Dank dem Hinweis in https://github.com/hbitter/DNMS/issues/21 bin ich nun an einem Setup, bei dem das Mikro weg vom SDS011 ist und per Kabel in eine Verteilerdose geführt wird. In der Bauanleitung werden 200mm Kabellänge genannt. Ab welcher Länge treten Probleme auf? LAN-Kabel wäre nun praktisch, weil schon genug adern in einem Kabel drin sind und ich nicht verschiedene Kabel zusammenschneiden und bündeln muss etc. An sich aber jammern auf hohem Niveau;)

hbitter, 2020-09-27:
Ab welcher Länge treten Probleme auf?
Gedacht ist I²S als eine Verbindung zwischen ICs auf einer Platine. Es wird sicher einiges mehr drin sein. Hängt aber stark vom Einzelfall und was man dann unternimmt z.B. abgeschirmte Kabel aber die ICs auf beiden Seiten müssen natürlich auch  entsprechend höhere Kapazitäten treiben zu können. Also??

closed 2020-10-30 by hbitter