hbitter/DNMS #26: Unterschied DNMS Versionen 1.0.4 zu 1.0.5?

erduino, 2020-10-26:
Hallo
Was ist eigentlich der Unterschied zwischen diesen 2 Versionen?

pe0ter, 2020-10-26:
Version 1.05 routes audio to the usb port for external recording/evaluation

erduino, 2020-10-26:
Okay, danke für die Info. 
Ich habe das Probleme das kaum 3 korrekte DNMS readings hintereinander funktionieren bis ich die Meldung "DNMS read failed" bekomme. Mit 1.04 ist es etwas besser aber auch nicht gut. Gibt es da vielleicht immer noch Probleme mit dem I2C Library?

---
End reading DNMS
Start reading DNMS
LAeq: 44.76
LA_MIN: 41.75
LA_MAX: 48.49
----
End reading DNMS
Start reading DNMS
DNMS read failed          <<<-----------------
----
End reading DNMS
Start reading DNMS
LAeq: 48.25
LA_MIN: 42.29
LA_MAX: 62.72

pe0ter, 2020-10-27:
Both my installed units work fine and the other eight other circuit board sets I have build and tested so far work flawless. I am using AirRohr 1.4 (NodeMCU V3) and DNMS 1.4 (1.04 firmware) circuit boards and a 3m RJ12 cable between them.

DNMS read failed could be a noisy I2C bus between NodeMCU and Teensy. 
What hardware setup are you using?

hbitter, 2020-10-27:
Hallo erduino,

1. Unterschied Firmware V1.0.4 und V1.0.5
Wie pe0ter schon sagte, wird bei der Firmware zusätzlich das Audio Signal (nach dem A-Filter) auf der USB Schnittstelle ausgegeben. Das Teensy Board ist am USB Port dann ein Audio Gerät (PCM 16-Bit, 44,1 kHz). Du kannst also das Signal weiterverwenden z.B. mittels Audacity aufzeichen und auswerten. Diese Funktion ändert an der restlichen Funktionalität nichts und hat (zu mindest theoretisch) keinen Einfluss auf die I²C Übertragung zwischen Teensy und NodeMCU.

2. I²C Verbindung zwischen Teensy und NodeMCU
Probleme mit der I²C Lib gibt es nicht mehr (bis jetzt sind mir jedenfalls keine bekannt). Um bei Deinem Problem weiter zu kommen, brauche ich die genaue Konfiguration HW, Firmware auf der NodeMCU, usw. Wenn Du eigene Boards einsetzt bitte auch ein Bild davon. Du kannst Das gerne auch per mail an mich schicken helmut.bitter (at) t-online.de.
Viele Grüße
Helmut

erduino, 2020-10-27:
Vielen Dank Helmut und Peter
Die detaillierte Konfiguration werde ich gerne schicken.
Bei der Gelegenheit wollte ich fragen ob es ein Quelle gibt wo man die fertigen aber unbestückten PCP's (circuit board ) beziehen kann oder hat hier jemand einen Restbestand zum Verkauf?

hbitter, 2020-10-27:
Hallo erduino,

der Kollege Franz Hoefle hat gerade einige Boards in China bestellt - sind gerade unterwegs. Hier seine Mail Adresse:
franz.hoefle (at) buergerforum-gladbeck.de
Viele Grüße
Helmut

pe0ter, 2020-10-27:
I ordered mine at PCBWay.com, 10 sets $38,00 including shipping, but they sent me 11 sets so I have 1 unused set available.

erduino, 2020-10-27:
> 
> 
> I ordered mine at PCBWay.com, 10 sets $38,00 including shipping, but they sent me 11 sets so I have 1 unused set available.

Ok, pe0ter, würdest du mir das übrig gebliebene Set verkaufen? In welchem Land wohnst du? Kannst mir sonst auch auf itas (at) gmx.ch schreiben.

erduino, 2020-10-27:
> 
> 
> Hallo erduino,
> 
> der Kollege Franz Hoefle hat gerade einige Boards in China bestellt - sind gerade unterwegs. Hier seine Mail Adresse:
> franz.hoefle (at) buergerforum-gladbeck.de
> Viele Grüße
> Helmut

Danke Helmut. Das dauerd ja wahrscheinlich noch einen Moment und es sieht so aus als hätte pe0ter noch eins übring.

pe0ter, 2020-10-27:
@erduino 
Ich Hab dir ein mail geschickt. Ich wohne in Holland, Eindhoven.

closed 2020-10-30 by hbitter