hbitter/DNMS #27: Bestückte Platinen bestellen - CPL File?

LaPhi, 2020-12-07:
Ich würde gerne das AIRROHR V1.4 bestückt bestellen wollen. 
Im Bestellprozess (JLCPCB) kommt die Frage nach einem CPL File (Pick&Place File,Only accept XLS,XLSX,CSV).
Existiert diese Datei in diesem Projekt? Wie könnte ich diese erzeugen?

pe0ter, 2020-12-07:
https://support.jlcpcb.com/article/84-how-to-generate-the-bom-and-centroid-file-from-kicad

Met vriendelijke groet,
Peter

hbitter, 2020-12-07:
Hallo LaPhi,

nein ein solcher CPL File für die Bestückung existiert bisher nicht. Entweder, wie Peter schon geantwortet hat, selbst erzeugen nach der Anleitung von JLCPCB oder wenn Du ein bestücktes Board kaufen möchtest den Kollegen Franz Hoefle "franz.hoefle (at) buergerforum-gladbeck.de"  fragen, der manchmal Boards vorrätig hat und auch bestückt.
Viele Grüße
Helmut

closed 2021-05-30 by hbitter