hbitter/DNMS #28: Question: why Teensy and not ESP32?

DeeKey, 2021-01-02:
Why to use expensive Teensy board and not ESP32 which also has LoRa support?
Is it theoretically possible to use ESP32 for combining both DNMS and AirRohr sensors? If yes then what is needed to implement this?

glosair, 2021-01-02:
The Teensy is used for it's raw processing power as the Mic is being monitored 24/7 to give the average sound level not point measurements that would be meaningless.

hbitter, 2021-01-02:
Hi DeeKey,

the selection of the Teensy Boards (3.6 and than 4.0) had mostly two reasons:

- availability of a good audio library for the Teensys, where all the necessary functions were available and fitting together. Now the firmware will be changed from 16-bit integer calculation to 32-bit floating point calculation. There is a special 32-bit floating point  version of the audio librar available, a great thing not to reinvent the wheel.
- the processing power of the Teensys compared to ESP32 is higher. Here is a comparison (from the Teensy developer Paul Stoffregen - PJRC:
[image]
This is especially important now going over to 32-bit floating point (which the ESP32 has as well in HW but with lower performance).

I don't think it is really a good idea to combine all the functions especially for DNMS and all the communication tasks on one microcontroller. Yes, theoretically it would be possible but it would loose simplicity and openness. We plan to do some frequency domain analysis in the future as well and therefore a lot of processing powers has to be available and as glosair said, noise measurement is a continous task not a spot measurement.
Hope this explains the motivation behind the decisions.
Regards,
Helmut

DeeKey, 2021-01-02:
Thank you for the comprehensive answer. It should definitely be added to FAQ section.
Let the issue be opened for now. So that others will read it as well.

DeeKey, 2021-01-02:
I was given a link which might pose an interest for this issue. Here is both code and prototype description of the sound meter based on ESP32 with the same type of microphone  ICS-43434. Maybe it be of interest...
https://github.com/ikostoski/esp32-i2s-slm

hbitter, 2021-01-02:
Hi DeeKey,

there more information abaout the ESP32 project of Ivan Kostoski on this [Hackaday page](https://hackaday.io/project/166867-esp32-i2s-slm). Also the limitations of the ESP32 are described when it comes to 64-bit floating point.
Regards,
Helmut

DeeKey, 2021-01-03:
Just for information: some details about performance of ESP32
https://www.esp32.com/viewtopic.php?f=14&t=800&sid=fb9d650ee6806b018767f48359b33d9d&start=20

closed 2023-02-11 by hbitter