hbitter/DNMS #29: Support of stereo microphone

DeeKey, 2021-01-06:
Is it make sense to support stereo microphone?
If yes then how to do it the best way? On the single PCB or using 2 microphone paired cases?

hbitter, 2021-01-06:
Hi DeeKey,

for ambient noise measurement I would say no. But if you have other purposes in mind why not. It will give you directional information.
The configuration of two microphones - where they are placed -  would also depend on the purpose.
A different story would be a microphone array, that could be used for localizing the source of noise.
Regards,
Helmut

DeeKey, 2021-01-06:
Does it make sense to install 2 microphone chips on a single PCB to make a redundant solution?
If one will fail (degrade) then the other one can still be used. 
The same approached is used in PurpleAir PM sensors. They install 2 PM sensors side by side. Thus if readings from both sensors differs too much then something is wrong.
Can the same approach be used for DNMS?

pe0ter, 2021-01-06:
Deekey,

Extending this redundancy line of thought, it might be easier to install 2 completely separate standard airRohr PM + DNMS meters as one can use the now readily available proven hardware and software and no recoding is required. Using 2 meters also ensures measurements continue to be registered with controller and power supply failure.

But as an environmentally interested experienced developer, you are free to build your own meter, using your own hardware, connect it to as many sensors you like and write your own code to publish measured data into the sensor.community IPA to make it available for the world. You can use the published github code for the standard airRohr controller and Teensy as a starter.

Met vriendelijke groet,
Peter

hbitter, 2021-01-06:
Hi DeeKey,

from the firmware point of view sure this can be done. It would nearly double the processing  load because all calculations would have to be done twice. I think with Teensy4 it should be possible. It has a lot of processing power.
If it would really help to detect degradation I would have doubts. Both microphones are sitting more or less close together. It means they are exposed to the same environmental conditions (weather ... temperature, humidity). Maybe the same will happen to both? Who knows?
Besides that there is always the question of casing of two microphones on a single PCB because the casing has a big impact of accuracy of the measurements.
As Peter suggested, using two systems would be a way as well.
Regards,
Helmut

closed 2021-05-30 by hbitter