hbitter/DNMS #33: what happened to type=Laerm

derkod, 2021-04-20:
Dear Helmuth,
Do you maybe understand why the SensorCommunity API to get the DNMS-data is not working for me?
See https://github.com/opendata-stuttgart/meta/issues/158
Thanks!
Derko

hbitter, 2021-04-20:
Hi Derko,
I don't know anything at the moment about the change that happend. I will ask around what happend and how to solve it. Hopefully I can come back in a short period and give more information.
Regards,
Helmut

ricki-z, 2021-04-21:
This should work again.
The correct URL for the request would be:
https://data.sensor.community/airrohr/v1/filter/country=NL&type=DNMS%20(Laerm)
The quotation marks were only written to show start and end of the query string.

closed 2021-05-30 by hbitter