hbitter/DNMS #34: Pesky ICS43434: Welche Komponenten neben dem ICS43434 sind auf dem PCB?

linuxlurak, 2021-06-05:
Leider finde ich keine BOM zum Pesky-Board auf tindie... Könnte mir jemand weiterhelfen?

Hier lese ich von power supply decoupling mittels 0.1uf capacitor und 100kΩ pull‐downresistor :
https://datasheet.octopart.com/ICS-43434-InvenSense-datasheet-62330883.pdf
[image]

Linker Baustein: Das dürfte der capacitor in der Nähe des 3V3 Pins sein.

[image]
(Quelle: https://oshpark.com/shared_projects/GiObJBkm)

Rechts davon dürfte der 100kΩ pull‐downresistor sein.

Ich komm nicht recht weiter beim Suchen der passenden Komponente... 

Für eine SKU zB bei Digi-Key oder Farnell wäre ich Euch sehr dankbar!

glosair, 2021-06-05:
Both are just 0805 parts, 

Part numbers from farnell 
    MCWR08X1003FTL
    MC0805B104K500CT

Or get a sample kit from china like this, you won't regret it.

https://www.ebay.co.uk/itm/164643706126?_trkparms=aid%3D1110006%26algo%3DHOMESPLICE.SIM%26ao%3D1%26asc%3D20201210111314%26meid%3D899ed7b4eb92492794d10fae4ec5b694%26pid%3D101195%26rk%3D2%26rkt%3D12%26sd%3D284118807552%26itm%3D164643706126%26pmt%3D1%26noa%3D0%26pg%3D2047675%26algv%3DSimplAMLv9PairwiseUnbiasedWeb%26brand%3DUnbranded&_trksid=p2047675.c101195.m1851&amdata=cksum%3A164643706126899ed7b4eb92492794d10fae4ec5b694%7Cenc%3AAQAFAAACEFM5LNacLo86I0%252FhZ3wCwQpzJwGT1DP4t4aqkvTu2mTAkVSaLURpiJvYSyZBd4dpyd7WrJUi4lrNNPT8qC3h4wbmzYAZDvv3gktUBy2EQ9NqEsN%252F8qdv9HbmVxu%252BbWkRih2wQWtyxYEE%252BMLoa%252BuQwJxOwtb9xW%252FzUnykn8QXD5CC24fk%252BMiHoqmDUfKiaoCy3uXeltbRZWWN7sk788WwrBz2GOtlJO43hPyrvCZtOcE890fhcIaRSFj%252FpP2UYUX0ZlAX7%252FW55A7EQRy7k6xixoLzf73kLOszz2Pd0X8cS9LfbUqiffegeIPlPmJWD4Xm2tjXyXQoYtYit8N8WSHuRV0gk9S6KsAt06wfh5y1ikaofZt9tFEgnfkW5u3T%252FlYy21AJnOGpACSFmH4ocb37so%252B7rN%252F6tw1h8%252BCh9d8hyq3c%252B6dE1gaQMqqeV1WlfL%252BB1M8tiVYKSc0GtXHv%252FL%252FbT%252BFygVkl4t%252BxKO%252BvyosY8ohMG4HpuDUqVo2rID7THIWI8aUUaZOzeSDqNXx%252BCoFSq1SxJ%252BlwCrIK9u%252BeckHIW93O7sStMSktSKsEjnk8w8uRQOi4997W%252FvxWDAclMAOXshFjHugYpCkXT9lfVVW7rZZsao3%252BW8UuysYjQ1jDhwUatxmj%252BhyYynm4BqVT47Mafncned3TTzMjI9DuyIVEVjjz4TP4TOnU5iHhmDZl7snLqg%253D%253D%7Campid%3APL_CLK%7Cclp%3A2047675

linuxlurak, 2021-06-05:
Thanks a lot @glosair !

I want to place an order on pcbway.com by uploading the boards gerber files and now i am in need of a bom. The partnumbers you provided help a lot :) Now I have to map the gerber with the bom, perhaps I have to rebuild the pcb design by my own.

Do you think this is a feasible way to get  microphone boards?

 An Order for 50 boards costs $90 dollars for the pcb and the assembling service. the microphone chip, resistor and cap not included. I guess a board will cost about 5 to 6 dollars in the end delivery included.

glosair, 2021-06-05:
Just realized they aren't 8805 sized they are 1608 so should be 
https://uk.farnell.com/yageo/ac0603fr-07100kl/res-100k-1-0-1w-0603-thick-film/dp/3495222?st=1608%20100k%20cut
and
https://uk.farnell.com/avx/06031c104k4z2a/cap-0-1uf-100v-mlcc-0603/dp/3521950?st=1608%200.1uf%20cut 

I had been looking at this and designed a board that would hold the mic directly onto a board attached to the teensy 4.0. the I2C driver was placed between the Teensey and a custom esp board.

Had problems with the production and the ground contact around the hole in the board and then found that the thickness of the board was effecting the levels massively.  

Spent some time talking to my brother inlaw (a sound engineer) and he suggested either the thinnest possible board or even better a flex pcb.  I had to put the project on hold as I've been building a lab and workshop at the end of the garden and it's taken up way too much time. 

[image]

glosair, 2021-06-05:
The pricing for the boards is a bit steep, you would be looking at panalising them allowing for wastage you should get 16-20 out of a 100mmx100mm PCB yes there is "engineering costs" but the PCBs come out as almost free

hbitter, 2021-06-06:
I made a layout for a round ICS-4343 Board that fits into 12,7mm plastic tube as well as in 13,0mm stainless steal tube. 

[image]

I have orderd assembled boards based on this layout twice in China (PCBWay). Did no panalising because I have no knowledge about that. Ordering 20 pieces including everything gave a price of about 10€ per piece at the end (incl. components, freight, Einfuhrumsatzsteuer). Going for 50 pieces or even more (200 would be a good number) would lower the cost.
If somebody is interested give me a personal mail (helmut.bitter (at) t-online.de.
Regards,
Helmut

glosair, 2021-06-06:
Afternoon Helmut

What thickness PCB are you using 0.4mm? 

Thanks
Tim

hbitter, 2021-06-06:
Hi Tim,

PCB is 1.6mm thick.
Regards,
Helmut

linuxlurak, 2021-06-06:
@hbitter wonderful pcb! 

Would it be ok for you to share the PCB and the BOM?

I could try my hand at panelizing. I have never ordered assembled PCB, but panelizing has worked. Could then make some calculations and if it makes sense, report about it and provide a few such pcbs also.

hbitter, 2021-06-06:
Hi linuxlurak,

there is absolutly no problem with sharing the PCB layout. Please give me your E-Mail adress, that I can send you all the information.
Reagrds,
Helmut

Fux2, 2021-09-06:
Hi Helmut,
what type of connector did you use in your ICS-43434 board?
I am looking for such a connector for another project.
How strong is it connected? Is it made for one time connection or can it be unplugged at least a few times without any damage at the pcb?
Thanks
Markus

hbitter, 2021-09-08:
Hi Markus,

connector is from JST, it's JST SH series.  
[Link to JST SH page](https://jst.de/product-family/show/65/sh)

I think, it can be unplugged a couple of times but the crucial point is the cable. 
A similar connector is used on the Sensirion SPS30 PM sensor JST ZH series (1.5mm pitch compared to 1.0mm pitch of SH series).  The following happened with the cable after plugging and unplugging it a couple of times:
[image]
It's more a problem of the cable. Was a cheap one bought via Aliexpress.

Regards,
Helmut

closed 2023-02-11 by hbitter