hbitter/DNMS #35: What do you think of this sensor Setup? (please refer to pictures attached)

linuxlurak, 2021-06-06:
For quite some time I have been working on a compact an easy to assemble design of an environment sensor. It uses an SDS011, BME280 and a DNMS. I'm dreaming of integrating a MultiGeiger (see https://github.com/ecocurious2/MultiGeiger). (Lack of time and know how at the moment to complete this idea...)

Please refer to the following pictures:
Internal arrangement:
[image]
View of BME280
[image]
Microphone arm using a goose neck to direct the microphone towards the noise source:
[image]
View A of soldered DNMS, here you can see pins soldered in addition to connect SDS011 (V5+GND directly under the ESP32):
[image]
View B of soldered DNMS, here you can see pins soldered in addition to connect SDS011 ((TX/RX oppsite side of DNMS-PCB):
[image]

What do you think of this setup (especially I'm interested in your opinion about the housing, microphone arm, positioning)? Criticism is very welcome!

hbitter, 2021-06-06:
Hi,

[image]

It looks a bit to close to the wall and the ceiling. Would a test be possible: place a second DNMS (or another SPL meter) for a comparision farther away from the wall and the ceiling and compare the measurements over 2 - 3 days. Reflections can easily make a difference of 3 - 5 dB(A) or even more but it has to be measured.
Regards,
Helmut

glosair, 2021-06-07:
Good Morning Helmut

Looks good

I would question the orientation of the SDS011.
[image]

You might want to drill a small drain hole in the back of the enclosure so any condensation will drain out that way instead of into either the BME280 or Mic. 

Regards 

Tim

linuxlurak, 2021-06-09:
I measured the distances and setup another sensor. Just this moment I had to pack things up for the moment as a heavy storm is coming up. But tomorrow I'll set up everything again. 
[image]

I measured the last couple of days. I'll post the links soon. Then we can compare the data.

linuxlurak, 2021-06-09:
The storm passed without hitting us... Weather prognosis was completely off; so sensor is set up again.

Link to Data of Sensor next to the wall (the one 18cm away from the wall and 38cm away from the ceiling with Spaghettiplatten):
https://maps.sensor.community/grafana/d/000000004/single-sensor-view-for-map?viewPanel=12&var-node=54519

Link to the other Sensor 80cm away from the wall and the ceiling:
https://maps.sensor.community/grafana/d/000000004/single-sensor-view-for-map?orgId=1&viewPanel=12&var-node=54520

I guess my setup is not very scientific... but i try my best.

linuxlurak, 2021-06-09:
> [image]
@glosair Thanks for this hint! I have to reflect on this fact some more. At the moment I did not think of any moisture accumulating in the case by the SDS011. And I used a small 3D-Printed cover with some small volume to prevent accumulating dust in the housing. 

The BME280 is in the middle mounted on a small round pcb that maches the whole. I think no condensation will drop on the bme280. most likely the condensation will be gathered in the 3d printed cover. Or do you have other experiences? In one of my sensors I mounted a moisture absorber silica gel in a simple casing/holder.

I inted to use the sensors on balconies. Perhaps I should measure the humidity in the housing too to check if there is a problem with condensation.

hbitter, 2021-06-09:
Hi linuxlurak,

do both DNMS running with the same firmware?

linuxlurak, 2021-06-10:
Actually I upgraded the firmware on all sensors with V2.0.5. But I did not check after the flashing procedure. Errors by me could be possible. I moved one of the sensor today to its former place and set up another one. Will share the links soon.

Do you have some indication there are two different firmwares used?

hbitter, 2021-06-10:
No, the question was only to be sure eleminating this point. 
Using firmware 2 will give LAeq values down to 30 dB(A) whereas firmware 1 will go down only up to 36/37 dBA. So you will see the difference only below 40 dB(A).
The differences between the two positions are really big even higher than expected, 1. picture at night at around 2:00:
[image]
Sensor 2 LAeq comes down to 35/36 dB(A), that is O.K.with firmware 1. So, will be interesting with firmware 2, if it's going down to around 30 dB(A).

2. picture at 14:00 Uhr
[image]
Nearly 15 dB(A) difference.

A cross check would be good, position sensor 1 micro close to micro of sensor 2 or the opposite (micro of sensor 2 close to micro of sensor 1)

Regards,
Helmut

linuxlurak, 2021-06-14:
DNMS next to the wall as before:
https://maps.sensor.community/grafana/d/000000004/single-sensor-view-for-map?viewPanel=12&var-node=54519

New measurements with new sensor:
https://maps.sensor.community/grafana/d/000000004/single-sensor-view-for-map?orgId=1&viewPanel=12&var-node=63125

I'm setting up a third DNMS to measure attached directly to the mic of the sensor next to the wall. Mmaybe i will only manage this next weekend.

linuxlurak, 2021-06-14:
An additional note on the installation of the SDS: The manual specifies to protect the SDS011 from light incidence. I have in the picture above in comment https://github.com/hbitter/DNMS/issues/35#issue-912905447 where the sensor is unscrewed, turned the by 90°. Better still is an opaque tube or a hose to take. In any case, the sensor opening must not be aligned against the opening in the housing. The result can be seen here, the impact is dramatic:
[image]

linuxlurak, 2021-07-28:
Now, during two or three weeks, the following set was measured:
[image]

The result seems to me far from the expected:
https://maps.sensor.community/grafana/d/000000004/single-sensor-view-for-map?viewPanel=12&var-node=54519&orgId=1
https://maps.sensor.community/grafana/d/000000004/single-sensor-view-for-map?orgId=1&viewPanel=12&var-node=63125

I'm lost...

closed 2023-02-11 by hbitter