hbitter/DNMS #36: Wrong DNMS-values since some days

imaluschi, 2021-06-08:
Strange behaviour of a DNMS (only DNMS). Worked quite a long time (since 09/2020) flawless, but since some days...
Any idea? Is something special with values around 46/47/48 (min, avs, max)?
Nothing was changed, the Sensor was not moved. Traffic on the street nearbay is nearly always the same (~12k cars/day)
[image]
[image]
https://maps.sensor.community/grafana/d/000000004/single-sensor-view-for-map?viewPanel=12&orgId=1&from=1621521113932&to=1623189943484&var-node=48547

Two other sensors (DNMS with SDS011 and BME280) still work more or less flawless (sometimes problems with BME, but see https://github.com/opendata-stuttgart/sensors-software/issues/739).

I have to pick up the sensor for investigation. Will give an update when investigated, but will take some days.

imaluschi, 2021-06-08:
Teensy 4.0, Firmware V1.0.5

hbitter, 2021-06-09:
Hi imaluschi,

it looks that it started on the 06/05. 
[image]
Was there heavy rain at that date?
I think the microphone is broken and you have to change it.
Regards,
Helmut

imaluschi, 2021-06-09:
Hi Helmut,

yes, last weeks it's raining from time to time, sometimes with thunderstorms.
Nevertheless the microphone should be dry. If wind is pressing water to the microphone, the water should drop down - microphone is mounted downwards in direction to the street (source of sound).
[image]
I'll give an update when I got the sensor and had a quick look.

Jürgen

linuxlurak, 2021-06-09:
(Sorry, a little offtopic)
Hi @imaluschi 

> https://maps.sensor.community/grafana/d/000000004/single-sensor-view-for-map?viewPanel=12&orgId=1&from=1621521113932&to=1623189943484&var-node=48547

Could you tell me how I get such a link for my dnms sensor? One of my sensors with dnms has ID esp8266-586719 (airrohr).

imaluschi, 2021-06-09:
Did you register the DNMS on https://devices.sensor.community/sensors/ ?
You can only register the PM-Sensor and the DNMS - so drop BME280 (it will work, but you will get warnings).
[image]

var-node="Sensor ID from the registered DNMS - see picture", in the shown case  "var-node=48506"
https://maps.sensor.community/grafana/d/000000004/single-sensor-view-for-map?viewPanel=12&orgId=1&from=1621521113932&to=1623189943484&var-node=xxxxx

From and to are the times, changeable. Zooming in helps to find the values. "to=now" for graph until now.

hbitter, 2021-06-09:
Hi,
I think, it's now possible to register more than 2 sensors for one NodeMCU ID. Rajko Zschiegner from Sensor.Community has done that change. Anyway if it's not activated in the self registering process, Rajko can register a third sensor manually in the database, you have to send him a mail (rajko@sensor.community).
Regards,
Helmut

linuxlurak, 2021-06-09:
@imaluschi Wow, superb! Thank you very much! Is this documented somewhere or is it common knowledge? If the latter, i think i will provide a commit to include this nice trick of yours in the according readme. This works like a charm: https://maps.sensor.community/grafana/d/000000004/single-sensor-view-for-map?viewPanel=12&orgId=1&from=1621521113932&to=1623189943484&var-node=54519

@hbitter Thanks for this hint! One of my newer sensors showed up with DNMS without special treatment by Rajko, it should work. Without the trick by imaluschi i get something like this:
[image]. No clue why or what the background there is. Seems something wrong with the scale.

imaluschi, 2021-06-10:
Helmut, I know about the discussion but it seems not that self-registering a third Sensor is possible at the moment. But anyway, it's working. PN-Sensor and DNMS sensor are the important ones to register, BMP/E or DHT will work without registering.
[image]

But as already mentioned, off topic ;-)

imaluschi, 2021-06-10:
It's a corrosion-Problem. Seems the Sensor is broken. Tried to lean it mechanically, no change.
In this case, I will change the noise-sensor unit.
[image]

hbitter, 2021-06-10:
I would say, it's looking ugly and broken.
Do you know what the black material is? Is it from the potting compound?
Regards,
Helmut

imaluschi, 2021-06-22:
I'm not really sure. It could be from the foam from the microphone windshield or biological.
Potting compound? Hmm, can't imagine.

closed 2023-02-11 by hbitter