hbitter/DNMS #37: build manual: silicone cable example part and distributor

Basseuph, 2021-06-15:
I was looking a bit around for getting the cables connecting the microphone PCB, so it might be a good idea to provide some more info about them. I now ordered `SiliVolt-E` cables of the manufacturer `Stäubli Electrical Connectors`, see [Bürklin Webshop](https://www.buerklin.com/de/Silikon-Schaltlitze-h%C3%B6chstflexibel-015-mm%C2%B2-Au%C3%9Fen-%C3%98-11-mm-rot/p/94F1079) as an example distributor. Maybe you can add this to the [build manual](https://sensor.community/de/sensors/dnms/). Thank you!

hbitter, 2021-06-15:
Hi Basseuph,
good idea. I will inform ghe people mantainingthe Sensor.Community page.
In my [build manual version 2.2 ](https://github.com/hbitter/DNMS/tree/master/Manual) there is on page 58 a link to Amazon for such cables.
[image]

[Amazon link](https://www.amazon.de/Flexibler-Silikondraht-Elektrische-Temperaturbest%C3%A4ndigkeit-verschiedene/dp/B07PVDWC5R/ref=psdc_2076369031_t3_B07TV54V2T)

Reagrds,
Helmut

Basseuph, 2021-06-15:
Hi Helmut,

ah I obviously missed that one. Maybe include the Amazon link in the description as well?

Anyway, thank you for your effort, especially the measurements, which show quite impressing results!

Regards,
Basseuph

Franz4596, 2021-06-20:
You get many suppliers für "silicon cabels AWG26" in ebay.de, amazon.de, aliexpress.com and electronic distributors (bürklin, conrad,...). 
But Silicon is not really neccessary, you can use an usual wire (finely stranded conductor, feindrähtige Litze), more important is a short length (<25cm).
franz

closed 2023-02-11 by hbitter