hbitter/DNMS #38: Airroht does not recognize DNMS

aircutter, 2021-07-18:
Hi,

not sure if this is the right place. I try anway.

I build a setup with Airrohr and Teensy 4.o according to the manaula on this page.

The Airrohr works great. The DNMS also with the USB testing software.

When I go now and combine the two the Webpage for the Sensor (Airrohr) says "Check wiring DMNS not found" (Similar to this).

I used different wires and even harwirde the connection. Seems not to be the problem. Falshing of the versions for Teensy, etc.

The only "abnormal" thing is that the Teensy LED (necx to the button) flashes with all versions. Maybe normal, maybe not ...

Now my question: HAs anybody a further idea how to debugg the situation and finding out what casues the comm issue (if it is a comm isssue)?

Thanks for your help!

Marc

hbitter, 2021-07-18:
Hi Marc,

- flashing of Teensy LED - yes, that is normal, flashing time is around 250 ms on and 250 ms off - that means firmware on Teensy is running
- following your description there must be a communication problem between NodeMCU and Teensy. Problem can be the cable or one of the I²C transceivers (P82B715)
- Cable - please recheck the connection - it may be that the cable is wrong and does not give a 1 : 1 connection for all 6 pins (an example - once the connector on one side of a bought cable was turned by 180°)
- P82B715 may be one is not soldered completely or may be one is dead after soldering?
- If you are not using the extended I²C connection between Airrohr and Teensy at all that is problem as well, because even if the connection is not too long, there is a problem running DNMS without the P82B715 transceivers on the I²C bus - it is not reliable (Look at my note about the combined version: With PCB versions DNMS-T4.0+NodeMCU-V1.x there is  ....) This would apply in the case you are not using the transceiver P82B715 on both sides.
I hope, it will help searching for the cause.
Regards,
Helmut

aircutter, 2021-07-18:
Hi Helmut,

 

many thanks for your insides!

 

I resoldered the P8s and that solved the problem. This time using a magnifying glass …

 

Measures values now! Nad they are showing in the community page.

 

Thanks!!

 

Marc

 

Hi Marc,
flashing of Teensy LED - yes, that is normal, flashing time is around 250 ms on and 250 ms off - that means firmware on Teensy is running
following your description there must be a communication problem between NodeMCU and Teensy. Problem can be the cable or one of the I²C transceivers (P82B715)
Cable - please recheck the connection - it may be that the cable is wrong and does not give a 1 : 1 connection for all 6 pins (an example - once the connector on one side of a bought cable was turned by 180°)
P82B715 may be one is not soldered completely or may be one is dead after soldering?
If you are not using the extended I²C connection between Airrohr and Teensy at all that is problem as well, because even if the connection is not too long, there is a problem running DNMS without the P82B715 transceivers on the I²C bus - it is not reliable (Look at my note about the combined version: With PCB versions DNMS-T4.0+NodeMCU-V1.x there is ....) This would apply in the case you are not using the transceiver P82B715 on both sides.
I hope, it will help searching for the cause.
Regards,
Helmut

closed 2023-02-11 by hbitter