hbitter/DNMS #39: Impossible DNMS Values on https://api-rrd.madavi.de/

terraformer, 2021-07-19:
Values on https://api-rrd.madavi.de/grafana/d/BYsfp-xGz/dnms?var-chipID=esp8266-9757414&orgId=1 are way off. On the Sensors Web interface everything looks good:

GET r: 200
output debug text to displays...
No newer version available.
.
WiFi connected, IP is: <confidential>
Starting Webserver... <confidential>

ChipId: 9757414

MAC Id: <confidential>
Read DNMS...
Trying DNMS sensor on 0x55ws: debug level ...
R/ DNMS
LAeq: 49.47
LA_MIN: 47.26
LA_MAX: 59.18
----
/R DNMS
## Sending to madavi.de: 
Succeeded - api-rrd.madavi.de
Time for Sending (ms): 241
R/ DNMS
LAeq: 49.14
LA_MIN: 46.96
LA_MAX: 60.33
----

Current Firmware: NRZ-2020-134-B3/DE (Apr 8 2021) 
Board: DNMS-T4.0+NodeMCU-V1.5

Any ideas how to fix this? Sensor seems ok but the values in grafana are impossible and do not match with the sensor log.

terraformer, 2021-07-21:
seems as if i put this issue on the wrong repo, so i reposted it here: https://github.com/opendata-stuttgart/sensors-software/issues/907 this one here can be closed i suppose.

hbitter, 2021-07-27:
Hi terraformer,

I was out of town for some days and found your question today. 
I think Pierre-Jean answered your qustion at Sensor.Community GitHub pages.
[Impossible DNMS Values on https://api-rrd.madavi.de/ #907](https://github.com/opendata-stuttgart/sensors-software/issues/907)

Getting some more reports on noise values have  [a look at this page](https://laerm.citysensor.de/l) from Reinhard Fürst (only in German).

Regards,
Helmut

closed 2023-02-11 by hbitter