hbitter/DNMS #47: Ideen für einen Wetterschutz (insbesondere Regen)

linuxlurak, 2022-01-01:
Gibt es erfahrungen mit "Wetterschutz" namentlich für Regen? Ich benutze für die Mikrofonplatinen im Moment einen Plastikprimer-Spray. Je nach Installation wird das Mikro bei Regen dann trotzdem Nass. 

Ein entsprechender Schaumstoffüberzug (auch gegen Wind) könnte da etwas den direkten REgen abfangen und das Wasser abtropfen lassen. Immerhin. Nach einem Halben jahr sind mir die Pesky-Mikroplatinen aber dann gestorben. Keine Messwerte mehr (nur eine 5.0 zB bei Laeq). Das Platinchen ist dann korrodiert, dort wo die Löstellen der Pins rausschauen (Bild folgt). 

Bald werde ich die neuen Platinen auf der Basis von hbitter's Design ausprobieren, die keine through hole Pinlöststellen mehr aufweisen. Mal schauen wie wetterfest die sind;)

Franz4596, 2022-01-01:
Ich verwende ein d~30mm l~45mm Muff Schaumstoff-Windschutz, der etwas 5...10mm nach vorne gezogen ist, so dass ein Hohlraum entsteht. Hatte auch ein Ausfall (esp8266-1492670, #38366) würde aber nicht auf Korrosion der Lötstellen vermuten. 
01/2021...01/2022 lief ein Persky/Tindy-breakout, dann deutlich niedriege SchallWerte. 
01/2022... läuft ein sensormaestros -breakout.
15um Silikonschutzfolie (USA) konnte nicht besorgt werden. 
Runde Breakouts ohne throuhhole pins habe ich auch eingesetzt und verfügbar (8,50€)

BasvanDrunen, 2022-01-01:
What I did is:I used the foam from a paint roller, picture attached. I mounted the mic downwards.Around the big plastic tube is a disc (not on the picture) preventing the rain to drip down from the tube into the foam.The foam has the function of wind cover as well as protection for the rain.The foam has minimal impact on the sound measurements.Best regards from the Netherlands, that is why I write in English, my German is not that good ;-)

BasvanDrunen, 2022-01-01:
My previous comment did not contain the images, that is why i add them now.

[image]
[image]

rin67630, 2022-05-16:
Sie können auf Seite 29 ff von mein Projekt lesen, wie ich das Problem gelöst habe:

Ein Garnelenfilter aus der Aquaristik und ein Edelstahltopfreiniger aus dem Supermarkt, das ist alles.

https://github.com/rin67630/Sound-pressure-level-meter-Booster/blob/master/EierlegendeWMS-Schallpegel.pdf

Kaum zu glauben, aber dadurch bleibt das Mikrofon bei Regen trocken, und Windgeräusche werden genauso gut transparent gedämpft, wie von prof. Winschutzhauben. 
Dazu ist alles 100% Edelstahl und verrottet nicht.

[image]

[image]

rin67630, 2022-05-16:
Ich habe noch eine Beschreibung meiner Lösung gefunden.

[DFLD-Outdoor-Mikrofon.pdf](https://github.com/hbitter/DNMS/files/8702005/DFLD-Outdoor-Mikrofon.pdf)

mullercomms, 2023-02-16:
Ein komplettes Kit aus ICS-43434 breakout board, Silikonkabeln, Cavity und Silikonfolie 0,1 mm zum Wetterschutz gemäss Invense Application note AN-1124 gibt es hier:
https://www.ebay.de/itm/363608386639