hbitter/DNMS #48: Fragwürdige Messergebnisse bei 200Hz - starke Schwankungen

imaluschi, 2023-02-11:
Andauernd sehr starke Schwankungen der Messergebnisse bei 200Hz. Da die Schwankungen auch bei gehobenen Pegel stattfinden und noch dazu die Schwankungsbreite ansteigt - das kann nicht sein.
Hier die Grafik aus Grafana, Werte werden alle 30 Sekunden zur InfluxDB gesendet. Um 6 Uhr läuten die Kirchenglocken, entsprechend der Pegelanstieg.

[image]

Kommt es hier irgendwo zu einer Signalstörung?

Konfiguration:
Lärmsensor mit Feinstaubsensor SDS011, DHT22 und BME280 (zum Vergleich). Lärmsensor ist etwa 1m von der NodeMCU-Einheit mit verbautem Feinstaubsensor (in Verteilerdose) entfernt, noch dazu "ums Eck".
Firmware Teensy 4.0: DNMS_V4.4.4

[image]

imaluschi, 2023-02-11:
Ergänzung:
Bei einem zweiten Lärmsensor OHNE Feinstaubsensor und Temperatur-/Feuchtemessung tritt dieses Phänomen nicht auf. Derzeitige Verwendung Indoor. Auch 30 Sekunden Intervall für Übertragung zur InfluxDB.

[image]
[image]

Franz4596, 2023-02-11:
Die auf der 200Hz Linie liegenden niedrigen Werte scheinen zusätzliche Messpunkte zu sein ?
Solche "Schwingungen" deuten evtl. auf Aliasingeffekten (Überlagerung) hin.

⁣VG Franz Hoefle

​

hbitter, 2023-02-11:
Hallo Jürgen

wie Franz schon bemerkt hat, gibt es auf der 200Hz Linie zusätzliche Messpunkte jeweils kurz vor dem richtigen Messpunkt im 30 Sekunden Abstand, wobei die zusätzlichen Messpunkte vom Abstand her zum richtigen Messpunkt schwanken, so wie ich es aus der Grafik sehe:
[image]
d.h. es werden zu dem 30 Sekunden Messintervall noch zusätzliche Werte, die zeitlich asynchron sind, nur für 200Hz dargestellt. Nimmt man nur die 200Hz Werte in des 30 Sekunden Rasters ergibt sich, so wie ich es sehe ein "vernünftiger" Verlauf.
Woher kommen die zusätzlichen Werte?
Ausschließen kann ich nichts auch nicht Beeinflussungen von den anderen Sensoren. Es kann aber auch an der InfluxDB und/oder Grafana liegen. Vorschlag für das weitere Vorgehen:

1. Die weiteren Sensoren an der NodeMCU nacheinander abklemmen. Gehen die Störungen/zusätzlichen Werte dann weg durch das Abklemmen eines Sensors?
2. Grafana Definitionen überprüfen, ist noch ein anderer Messwert definiert, der mit der gleichen Farbe wie der LAeq200z angezeigt wird?
3. InfluxDB: Sind in der Messreihe unter dem gleichen Namen für den LAeq200Hz diese zusätzlichen Messwerte abgelegt?

Viele Grüße
Helmut

imaluschi, 2023-02-12:
Auweh, peinlich! Habs gefunden. Hab bei den Grafana-Einstellungen bei den 200Hz vergessen den Node anzugeben.
Habe derzeit zwei Sensoren laufen, einen Indoor einen Outdoor, selbes Intervall mit dem zur InfluxDB gesendet wird - die können  natürlich praktisch nie synchron laufen. Durch die fehlende Angabe des Nodes hat Grafana auch die Werte des Indoor-DNMS mit angezeigt, die natürlich niedriger sind als die Outdoor-Werte. Daher die Schwankungen.
Danke für die Hinweise. Darauf, dass Werte zeitlich leicht versetzt waren, bin ich in meiner Hudelei nicht gekommen.

[image]
[image]

Damit werden nun auch die Werte von gestern (Bild im Ausgangspost) richtig dargestellt.

[image]

LG Jürgen

closed 2023-02-12 by imaluschi