hbitter/DNMS #49: Arduino IDE compiling error

tk5ep, 2023-04-04:
Hi,

I try to compile your sketch with the Arduino IDE and get several errors like this :

_C:\Users\Admin\Documents\Arduino\Datas ARDUINO\NRZ-2020-134-DNMS-4.4\airrohr-firmware\airrohr-firmware.ino:3993:24: error: 'void fetchSensorPMS(String&)' was declared 'extern' and later 'static' [-fpermissive]
 3993 | static __noinline void fetchSensorPMS(String &s)_

What should I do ?

Thanks,

hbitter, 2023-04-04:
Hi,
For the Arduino IDE a file with the name platform.local.txt has to be generated and placed at the right position with the following content:

compiler.c.extra_flags=-fpermissive
compiler.cpp.extra_flags=-fpermissive

The location of the file platform.local.txt depends on the OS you are using as well on the version of the Arduino IDE you are using. 

Hope this helps, please let me know.
Regards,
Helmut

tk5ep, 2023-04-04:
Hi, and thanks for fast anwers.
To find he right location, I've made the compiler verbose and on top of the compiler output, I have :

_Using board 'nodemcuv2' from platform in folder: C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2
Using core 'esp8266' from platform in folder: C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2_

Si, I guess, it's the right place where I found a platform.txt file.
I've added the suggested platform.local.txt, with the following lines :

compiler.c.extra_flags=-fpermissive
compiler.cpp.extra_flags=-fpermissive

But unfortunately, I have the same errors.
I'm using IDE 2.0.4 and Windows 10.

Thanks,
Patrick

tk5ep, 2023-04-04:
Helmut,
I had to restart the IDE to get better results ! 
It seems that the errors are now warnings, but I still have an error at the end. 

Most make reference to :

undefined reference to `_ZN17LiquidCrystal_I2C9setCursorEhh
undefined reference to `_ZN11TinyGPSPlusC1Ev'

Perhaps a library version problem ?
What libraries must be used for tinygps and liquidcrystal ?

Here what I get at compiler end :

Using library LittleFS at version 0.1.0 in folder: C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\LittleFS 
Using library ESP8266HTTPClient at version 1.2 in folder: C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\ESP8266HTTPClient 
Using library ESP8266WiFi at version 1.0 in folder: C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\ESP8266WiFi 
Using library ESP8266WebServer at version 1.0 in folder: C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\ESP8266WebServer 
Using library ESP8266mDNS at version 1.2 in folder: C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\ESP8266mDNS 
Using library EspSoftwareSerial at version 8.0.1 in folder: C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\SoftwareSerial 
Using library Hash at version 1.0 in folder: C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\Hash 
Using library ESP8266 and ESP32 OLED driver for SSD1306 displays at version 4.3.0 in folder: C:\Users\Admin\Documents\Arduino\libraries\ESP8266_and_ESP32_OLED_driver_for_SSD1306_displays 
Using library Wire at version 1.0 in folder: C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\Wire 
Using library ArduinoJson at version 6.21.1 in folder: C:\Users\Admin\Documents\Arduino\libraries\ArduinoJson 
Using library DNSServer at version 1.1.1 in folder: C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\DNSServer 
Using library Adafruit HTU21DF Library at version 1.1.0 in folder: C:\Users\Admin\Documents\Arduino\libraries\Adafruit_HTU21DF_Library 
Using library Adafruit BusIO at version 1.14.1 in folder: C:\Users\Admin\Documents\Arduino\libraries\Adafruit_BusIO 
Using library SPI at version 1.0 in folder: C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\libraries\SPI 
Using library Adafruit BMP085 Library at version 1.2.2 in folder: C:\Users\Admin\Documents\Arduino\libraries\Adafruit_BMP085_Library 
Using library Adafruit SHT31 Library at version 2.2.0 in folder: C:\Users\Admin\Documents\Arduino\libraries\Adafruit_SHT31_Library 
Using library DallasTemperature at version 3.9.0 in folder: C:\Users\Admin\Documents\Arduino\libraries\DallasTemperature 
Using library OneWire at version 2.3.7 in folder: C:\Users\Admin\Documents\Arduino\libraries\OneWire 
Using library SparkFun SCD30 Arduino Library at version 1.0.20 in folder: C:\Users\Admin\Documents\Arduino\libraries\SparkFun_SCD30_Arduino_Library 
Using library SparkFun SCD4x Arduino Library at version 1.0.4 in folder: C:\Users\Admin\Documents\Arduino\libraries\SparkFun_SCD4x_Arduino_Library 
Using library Sensirion I2C SEN5X at version 0.3.0 in folder: C:\Users\Admin\Documents\Arduino\libraries\Sensirion_I2C_SEN5X 
Using library Sensirion Core at version 0.6.0 in folder: C:\Users\Admin\Documents\Arduino\libraries\Sensirion_Core 
exit status 1

Compilation error: exit status 1

regards,,
Patrick

hbitter, 2023-04-04:
Hi Patrick,

TinyGPSPlus is: Version 1.0.3 TinyGPSPLus by Mikal Hart
LiquidCrystal is: Version 1.1.2 LiquidCrystal I2C by Frank de Brabander

Regards,
Helmut

tk5ep, 2023-04-04:
Helmut,

There are sooooo many different libraries for the Tinygps and liquidcrystal !

I installed the right TinyGPSplus library  and already had to good one for liquidcrystal. I had to force the use of the good one, as I have 2 different installed.

Now the tinygps errors are gone, but I still have the liquidcrystal errors showing up !

`c:/users/admin/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\Admin\AppData\Local\Temp\arduino\sketches\29B0CD124A6AD412BD17D1023BAE93C1\sketch\airrohr-firmware.ino.cpp.o: in function `IPS_version_date':
C:\Users\Admin\Documents\Arduino\Datas ARDUINO\NRZ-2020-134-DNMS-4.4\airrohr-firmware/airrohr-firmware.ino:1164: undefined reference to `_ZN17LiquidCrystal_I2C5clearEv'
c:/users/admin/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\Admin\AppData\Local\Temp\arduino\sketches\29B0CD124A6AD412BD17D1023BAE93C1\sketch\airrohr-firmware.ino.cpp.o: in function `_ZplIA2_cvE6StringRKS1_RKT_':
C:\Users\Admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.2\cores\esp8266/WString.h:413: undefined reference to `_ZN17LiquidCrystal_I2C9setCursorEhh'
c:/users/admin/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\Admin\AppData\Local\Temp\arduino\sketches\29B0CD124A6AD412BD17D1023BAE93C1\sketch\airrohr-firmware.ino.cpp.o: in function `_ZL13display_debugRK6StringS1_':
C:\Users\Admin\Documents\Arduino\Datas ARDUINO\NRZ-2020-134-DNMS-4.4\airrohr-firmware/airrohr-firmware.ino:782: undefined reference to `_ZN17LiquidCrystal_I2C5clearEv'
c:/users/admin/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\Admin\Documents\Arduino\Datas ARDUINO\NRZ-2020-134-DNMS-4.4\airrohr-firmware/airrohr-firmware.ino:783: undefined reference to `_ZN17LiquidCrystal_I2C9setCursorEhh'
c:/users/admin/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\Admin\Documents\Arduino\Datas ARDUINO\NRZ-2020-134-DNMS-4.4\airrohr-firmware/airrohr-firmware.ino:786: undefined reference to `_ZN17LiquidCrystal_I2C9setCursorEhh'
c:/users/admin/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\Admin\Documents\Arduino\Datas ARDUINO\NRZ-2020-134-DNMS-4.4\airrohr-firmware/airrohr-firmware.ino:790: undefined reference to `_ZN17LiquidCrystal_I2C5clearEv'
c:/users/admin/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\Admin\Documents\Arduino\Datas ARDUINO\NRZ-2020-134-DNMS-4.4\airrohr-firmware/airrohr-firmware.ino:791: undefined reference to `_ZN17LiquidCrystal_I2C9setCursorEhh'
c:/users/admin/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\Admin\Documents\Arduino\Datas ARDUINO\NRZ-2020-134-DNMS-4.4\airrohr-firmware/airrohr-firmware.ino:793: undefined reference to `_ZN17LiquidCrystal_I2C9setCursorEhh'
c:/users/admin/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\Admin\AppData\Local\Temp\arduino\sketches\29B0CD124A6AD412BD17D1023BAE93C1\sketch\airrohr-firmware.ino.cpp.o: in function `display_values':
C:\Users\Admin\Documents\Arduino\Datas ARDUINO\NRZ-2020-134-DNMS-4.4\airrohr-firmware/airrohr-firmware.ino:5783: undefined reference to `_ZN17LiquidCrystal_I2C5clearEv'
c:/users/admin/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\Admin\Documents\Arduino\Datas ARDUINO\NRZ-2020-134-DNMS-4.4\airrohr-firmware/airrohr-firmware.ino:5783: undefined reference to `_ZN17LiquidCrystal_I2C9setCursorEhh'
c:/users/admin/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\Admin\Documents\Arduino\Datas ARDUINO\NRZ-2020-134-DNMS-4.4\airrohr-firmware/airrohr-firmware.ino:5783: undefined reference to `_ZN17LiquidCrystal_I2C9setCursorEhh'
c:/users/admin/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\Admin\Documents\Arduino\Datas ARDUINO\NRZ-2020-134-DNMS-4.4\airrohr-firmware/airrohr-firmware.ino:5783: undefined reference to `_ZN17LiquidCrystal_I2C9setCursorEhh'
c:/users/admin/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.1.0-gcc10.3-e5f9fec/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\Admin\AppData\Local\Temp\arduino\sketches\29B0CD124A6AD412BD [... cut ...] p.o: in function `display_values':
C:\Users\Admin\Documents\Arduino\Datas ARDUINO\NRZ-2020-134-DNMS-4.4\airrohr-firmware/airrohr-firmware.ino:5997: undefined reference to `_ZN17LiquidCrystal_I2C9backlightEv'
collect2.exe: error: ld returned 1 exit status`

Any idea what else I can do ?

BTW, thanks for the time you take !
Meanwhile, I scanned thru your folders and I must compliment you for the fantastic work you've done. The manual is amazing !
Fortunately, I read and speak German.  :-)

Patrick

hbitter, 2023-04-04:
Hi Patrick,

maybe it helps to search within Arduino IDE library manager for liquidcrystal:

[image]

Regards,
Helmut

tk5ep, 2023-04-04:
Success !

I had to put the liquidcrystal library in the sketch folder and load it with 
`#include <LiquidCrystal_I2C.h"`

Now I have to find out how this works. :-)
I opened the terminal and at 9600 bauds, I can see that it works.

I can connect to the AP. I must switch to the English version.. LOL

Is there an operating manual somewhere ?
My main interest is the small particles measurement. Is there a MQTT implemented ?

Again, thanks a million for the help.

Sincerely,
Patrick.

hbitter, 2023-04-04:
Hi Patrick,

good that is works now :-)
MQTT: no, it is not implemeted yet. What we are using espacially for noise data is an InfluxDB (but only version 1.8 is supported up to now!)
Manual: no - but you will find some more information at Sensor.Community [https://sensor.community/de/](url) where the basis is from. My main interest is noise measurement, still I have added the last Sensirion Sen5X PM sensors to the system.

Regards,
Helmut

tk5ep, 2023-04-04:
OK Helmut,

Do you think it would be difficult to implement MQTT ?

In the code, I see that there is a parameter SEND2MQTT in ext_def.h.
But like all other parameters SEND2xxx it is set to 0.

In the main sketch I see that an MQTT  info :
`rejected (see issue #33)`

Regards,
Patrick.

hbitter, 2023-04-04:
Hi Patrick,
difficult to implement MQTT ? It depends on knowledge and how much time is one willing to spend.

SEND2MQTT in ext_def.h. But like all other parameters SEND2xxx it is set to 0.
These  defines give the default state for the Config Webpage, wghere you can set what APIs should be used for transmisssion of data.

For getting more info about the plans to add MQTT it would be better to ask the Sensor.Community poeple.

Regards,
Helmut

Ursu-Alex, 2024-05-30:
I have a similar issue with : WARNING: library LiquidCrystal I2C claims to run on avr architecture(s) and may be incompatible with your current board which runs on esp32 architecture(s).
In file included from C:\Users\alexu\Documents\Arduino\libraries\ESP8266_Influxdb\src\InfluxDbClient.h:31,
                 from C:\Users\alexu\Documents\Arduino\libraries\ESP8266_Influxdb\src\InfluxDbClient.cpp:27:
C:\Users\alexu\Documents\Arduino\libraries\ESP8266_Influxdb\src\HTTPService.h:87:5: error: 'WiFiClient' does not name a type
   87 |     WiFiClient *_wifiClient = nullptr;
      |     ^~~~~~~~~~
In file included from C:\Users\alexu\Documents\Arduino\libraries\ESP8266_Influxdb\src\InfluxDbClient.h:31,
                 from C:\Users\alexu\Documents\Arduino\libraries\ESP8266_Influxdb\src\InfluxData.h:27,
                 from C:\Users\alexu\Documents\Arduino\libraries\ESP8266_Influxdb\src\InfluxDb.h:30,
                 from C:\Users\alexu\Documents\Arduino\libraries\ESP8266_Influxdb\src\InfluxDb.cpp:27:
C:\Users\alexu\Documents\Arduino\libraries\ESP8266_Influxdb\src\HTTPService.h:87:5: error: 'WiFiClient' does not name a type
   87 |     WiFiClient *_wifiClient = nullptr;
      |     ^~~~~~~~~~
In file included from C:\Users\alexu\Documents\Arduino\libraries\ESP8266_Influxdb\src\BucketsClient.h:30,
                 from C:\Users\alexu\Documents\Arduino\libraries\ESP8266_Influxdb\src\BucketsClient.cpp:27:
C:\Users\alexu\Documents\Arduino\libraries\ESP8266_Influxdb\src/HTTPService.h:87:5: error: 'WiFiClient' does not name a type
   87 |     WiFiClient *_wifiClient = nullptr;
      |     ^~~~~~~~~~
In file included from C:\Users\alexu\Documents\Arduino\libraries\ESP8266_Influxdb\src\HTTPService.cpp:2:
C:\Users\alexu\Documents\Arduino\libraries\ESP8266_Influxdb\src\HTTPService.h:87:5: error: 'WiFiClient' does not name a type
   87 |     WiFiClient *_wifiClient = nullptr;
      |     ^~~~~~~~~~
C:\Users\alexu\Documents\Arduino\libraries\ESP8266_Influxdb\src\HTTPService.cpp: In constructor 'HTTPService::HTTPService(ConnectionInfo*)':
C:\Users\alexu\Documents\Arduino\libraries\ESP8266_Influxdb\src\HTTPService.cpp:37:5: error: 'WiFiClientSecure' was not declared in this scope; did you mean 'NetworkClientSecure'?
   37 |     WiFiClientSecure *wifiClientSec = new WiFiClientSecure;
      |     ^~~~~~~~~~~~~~~~
      |     NetworkClientSecure
C:\Users\alexu\Documents\Arduino\libraries\ESP8266_Influxdb\src\HTTPService.cpp:37:23: error: 'wifiClientSec' was not declared in this scope
   37 |     WiFiClientSecure *wifiClientSec = new WiFiClientSecure;
      |                       ^~~~~~~~~~~~~
C:\Users\alexu\Documents\Arduino\libraries\ESP8266_Influxdb\src\HTTPService.cpp:37:43: error: 'WiFiClientSecure' does not name a type; did you mean 'NetworkClientSecure'?
   37 |     WiFiClientSecure *wifiClientSec = new WiFiClientSecure;
      |                                           ^~~~~~~~~~~~~~~~
      |                                           NetworkClientSecure
C:\Users\alexu\Documents\Arduino\libraries\ESP8266_Influxdb\src\HTTPService.cpp:47:5: error: '_wifiClient' was not declared in this scope
   47 |     _wifiClient = wifiClientSec;
      |     ^~~~~~~~~~~
C:\Users\alexu\Documents\Arduino\libraries\ESP8266_Influxdb\src\HTTPService.cpp:49:5: error: '_wifiClient' was not declared in this scope
   49 |     _wifiClient = new WiFiClient;
      |     ^~~~~~~~~~~
C:\Users\alexu\Documents\Arduino\libraries\ESP8266_Influxdb\src\HTTPService.cpp:49:23: error: 'WiFiClient' does not name a type
   49 |     _wifiClient = new WiFiClient;
      |                       ^~~~~~~~~~
C:\Users\alexu\Documents\Arduino\libraries\ESP8266_Influxdb\src\HTTPService.cpp: In destructor 'HTTPService::~HTTPService()':
C:\Users\alexu\Documents\Arduino\libraries\ESP8266_Influxdb\src\HTTPService.cpp:64:6: error: '_wifiClient' was not declared in this scope
   64 |   if(_wifiClient) {
      |      ^~~~~~~~~~~
C:\Users\alexu\Documents\Arduino\libraries\ESP8266_Influxdb\src\HTTPService.cpp:65:12: error: type '<type error>' argument given to 'delete', expected pointer
   65 |     delete _wifiClient;
      |            ^~~~~~~~~~~
C:\Users\alexu\Documents\Arduino\libraries\ESP8266_Influxdb\src\HTTPService.cpp: In member function 'bool HTTPService::beforeRequest(const char*)':
C:\Users\alexu\Documents\Arduino\libraries\ESP8266_Influxdb\src\HTTPService.cpp:135:28: error: '_wifiClient' was not declared in this scope
  135 |    if(!_httpClient->begin(*_wifiClient, url)) {
      |                            ^~~~~~~~~~~
In file included from C:\Users\alexu\Documents\Arduino\libraries\ESP8266_Influxdb\src\InfluxDbClient.h:31,
                 from C:\Users\alexu\Documents\Arduino\libraries\ESP8266_Influxdb\src\InfluxData.h:27,
                 from C:\Users\alexu\Documents\Arduino\libraries\ESP8266_Influxdb\src\InfluxData.cpp:27:
C:\Users\alexu\Documents\Arduino\libraries\ESP8266_Influxdb\src\HTTPService.h:87:5: error: 'WiFiClient' does not name a type
   87 |     WiFiClient *_wifiClient = nullptr;
      |     ^~~~~~~~~~

exit status 1

Compilation error: exit status 1

hbitter, 2024-05-31:
Hi Ursu-Alex,

you wrote: "I have a similar issue with with your current board which runs on esp32 architecture(s)."

Does it mean you try to compile for ESP32???

If yes, that is not supported from my side. The Airrohr-Board is an Board for NodeMCU ESP8266. The NodeMCU firmware can run on other ESP866 systems as well (Wemos e.g.). In the firmware there are a couple of Ifdefs for ESP32, because the whole firmware is a fork of the Sensor,community firmware with specific additions for DNMS.

The Sensor.Community firmware does support basic DNMS features but not the full functionality. I don't know if there is a ESP32 version available from Sensor.Community.

If you try to compile for ESP8266 let me know, so that we can try to identify the problem.

Regards,
Helmut

Ursu-Alex, 2024-05-31:
Hi , so i run the code on a ESP32 but the same code works 2 days ago and now just don't ....

hbitter, 2024-05-31:
Hi Ursu-Alex,
I don't fully understand your answer. Do you mean, you could compile the code 2 days ago for an ESP32 and now compilation gives the errors. I didn't update GitHub within the last 2 days. 

Question is, how can I help you?

Regards,
Helmut

Ursu-Alex, 2024-05-31:
Hi , yes , 2 days ago the same code was working , i didn't change anything and now it doesn't work . Even the examples from library don't work .

hbitter, 2024-05-31:
Hi Ursu-Alex,

would you mind to give some more details:
- what HW are you using?
- Arduino IDE version, which Board is selected, what is selected under Tools (screenshots e.g)?
- which examples from which library?

Regards,
Helmut

Ursu-Alex, 2024-05-31:
[image]

hbitter, 2024-05-31:
Hi Ursu-Alex,

you are trying to compile for ESP32 an example (probably BasicWrite.ino from https://github.com/tobiasschuerg/InfluxDB-Client-for-Arduino) as far as I can see.
Sorry, but I cannot help. I have experience with that libraray and it is in no way linked to DNMS.

Regards,
Helmut

closed 2024-06-06 by hbitter