hbitter/DNMS #50: No API's in config NRZ-2021-134-DNMS-4.4/EN

WoffieZot, 2023-04-21:
I dont see any possibillity for api config on the the api's config page
[image]

WoffieZot, 2023-04-21:
Seems to work in Edge but not in Chrome, what could this be?

hbitter, 2023-04-21:
Hi,
exactly what you discovered happened to me during development as well. I thought it was cured with switching to esp8266 lib 3.1.1. I have to check again and will come back. Hopefully you can do your configuration in the meantime with Edge.
Regards,
Helmut

WoffieZot, 2023-04-21:
It is not a very big issue, i can live with it.

hbitter, 2023-07-21:
In the meantime I tried to reproduce this problem you had with Version NRZ-2021-134-DNMS-4.4/EN  on 5 different NodeMCUs but without success. Hopefully the new version NRZ-2020-134-DNMS-5.2 and NRZ-2020-134-DNMS-5.2-en will not have this problem.
Regards,
Helmut