hbitter/DNMS #51: Meassuring interval and data retreival

WoffieZot, 2023-04-21:
Question about the meassuring intreval. Standard it is set to 145sec, but it seems, i cannot make it shorter.
Ans also, the time is 145 sec, does this mean that the values LAeq, LAmax and LAmax are the average values over this set time. In this case the average over 145 sec?
Furthermore, i like the frequency spectrum values as well and i am looking to retrieve these values and seems that madavi doesnt store them, also no sound data comes vrom madavi? How to retreive all sound data, is there an api to get them from somewhere?

[image]

hbitter, 2023-04-21:
Hi WoffieZot,

- measuring interval
  Sensor.Community and Madavi only support only 145 - 150 seconds as measurement interval. So, if you check transmission to Sensor.Community and Madavi the value for "Measuring interval" is set to 145 sec. If you don' t check Sensor.Community and Madavi you can sat it as you like, depending on your needs from 1 sec to 3600 sec.
- LAeq, LAmax and LAmax are the average values ?
  Yes, it is the Time-Average Sound Level or Equivalent Continuous Sound Level over the choosen measuring interval (again from 1 sec to 3600 sec).
- frequency spectrum
  Frequency spectrum values are not supported by Senso.Community an Madavi up to now. The values are shown locally on the webpage but also transmitted to further APIs e.g. to an influxDB (version 1.8). Therefor we also introduced a 2nd measurment interval, which is independent from the Sensor.Community interval. You can set it to a short interval and transmit those values to e.g. influxDB. Grafana can be used for visualizing the data.
[image]

Regards,
Helmut

WoffieZot, 2023-04-22:
Do you know how to retreive histrorical data from madavi? I can only export to csv from their website by hand and that is only the last 24h.

WoffieZot, 2023-04-22:
The madavi data doesnt make any sense:
[image]

hbitter, 2023-04-22:
Hi WoffieZot,

- Historical data
access to historical data of Sensor.Community is via this Link: [https://archive.sensor.community/](url)
There you can retrieve all historical data from Sensor.Community on daily basis for each sensor by sensor ID (this is not the NodeMCU-ID but the additional ID each individual sensor gets after registration by Sensor.Community). 
Have a look at this site: [https://laerm.citysensor.de/](url). The values are retrieved from Sensor.Community.

- Madavi data
  Problem is known by Sensor.Community people. They gave answers to that question a couple of times in there forum. Please, have a look there.

Regards,
Helmut

closed 2023-07-21 by hbitter