hbitter/DNMS #54: No negative dB correction possible?

pe0ter, 2024-04-13:
Today we tested one of my ICS4-43434 microphones and one of my IM72D128 microphones using a B&K calibration unit at 94dB.
Results were a +1.6dB correction requirement for the ICS43434 unit and a -0.7dB correction for the IM72D128 unit.
Entering the +1.6dB correction on the ICS43434 unit worked great but entering a -0.7dB on the IM72D128 resulted in a +0.7dB correction. 
Could it be the error correction formula can not handle negative correction values?

hbitter, 2024-04-13:
Hi pe0ter,

I will check and come back. It may take some days.

Regards,

Helmut

hbitter, 2024-04-13:
Hi pe0ter,

just to confirm, there is a bug with small negative values in the range -0.0 to -0.999. Negative values from -1.0 seem to be ok. I need to look into it and prepare a fix. This will take a few days.

Regards,

Helmut

hbitter, 2024-04-15:
Hi pe0ter,

I just uploaded a new version of the NodeMCU firmware (NRZ-2020-134-DNMS-5.6) fixing the bug.

Regrads,

Helmut

closed 2024-04-28 by hbitter