hbitter/DNMS #56: English translation of DNMS manual

mrrco-gh, 2024-09-05:
Hello, while building two DNM sensord, I am translating 2.2 manual to English - using Google and a friend help because I do not know German.
To contribute to project, I wish to add my translation to repository: do you think it will be useful ?

hbitter, 2024-09-05:
Hi mrrco-gh,

This is a very nice offer. Sure, the manual itself is a bit outdated, but it is still worth translating to get all the bells and whistles. If you would do the translation, I would add it to the repository. Thanks a lot 

Helmut

marteinnm, 2025-05-21:
Hey @mrrco-gh,
Did you ever get through the whole manual? I am initiating a project using the DNMS and am trying to read through the manual but my deutsch isn't that great.