hbitter/DNMS #57: Not an issue, just a question

mark-joe, 2024-12-17:
Hello all,

I see a Raspberry software section that interfaces with a MEMS microphone, but I do not see a connection diagram. Do you connect Raspberry direct to the DNMS board (with the Teensy) or is it done differently? Thanks!

hbitter, 2024-12-17:
Hi mark-joe,

the connection between the Raspberry and the DNMS Teensy board is the same as the connection between the NodeMCU and the DNMS Teensy board. Basically it is an I²C connection. I²C is limited in cable length, so we use an extended version that allows a cable length of 10m between the two systems. You can either make a direct short I²C connection between the Raspberry and the DNMS Teensy board or use the extended I²C connection. We have provided a small Raspberry DNMS-Hat [https://github.com/hbitter/DNMS/tree/master/PCBs/DNMS-RPI-HAT](https://github.com/hbitter/DNMS/tree/master/PCBs/DNMS-RPI-HAT) that implements the extended I²C connection for the Raspberry.

Regards,
Helmut

mark-joe, 2024-12-17:
Hi Helmut

Thank you for your prompt reply, *much appreciated*. I now have a sound measuring system with a ICS-43434, but the  IM72D128V01 seems attractive as it is waterproof.  Also replacing a microcontroller with a raspberry is attractive as I just can login and update the code (otherwise I have to climb on the roof and get the microcontroller, flash it, and return it to the roof). Well, I'll see what it will be. Thanks again! Mark

closed 2024-12-17 by mark-joe