hbitter/DNMS #59: Frage zu I2C Decibel Sound Level Meter

klacol, 2025-05-03:
Hallo, ihr kennt euch hier anscheinend sehr gut aus mit Lärmmessung. Ich wohne an einer lauten Straße und interessiere mich daher auch dafür.

Ich habe mir diesen Sensor aus Indien bestellt: https://pcbartists.com/product/i2c-decibel-sound-level-meter-module/

War mit ca. 60 Euro gar nicht mal so preiswert, finde ich. Ich habe die Version ``Spectrum-Analyser`` an einen ESP32 angeschlossen und publiziere die Messwerte via MQTT in mein Home Assistant. Die Varianten ``PRO`` habe ich auch gekauft, aber noch nicht ausprobiert.

Ich kenne mich mit Lärmmessung überhaupt nicht aus. Was sagt ihr als Fachleute zu den Sensoren? Ihr habe ja dieses Rohr, das ist anscheinend wichtig, oder?

hbitter, 2025-05-07:
Hallo klacol,

den von Dir erwähnten Sensor kenne ich nicht und kann deshalb zu der Qualität nichts sagen. Wenn ich mir die Webseite ansehe und die verschiedenen Versionen vergleiche, kann ich folgendes sagen:
- Frequenzbereich ist bei allen Versionen eingeschränkt (PRO 20 - 8000 Hz) - sicher geht der Bereich oberhalb 10000 - 12500 Hz bei Umweltlärm nicht besonders ein aber 8000 Hz ist vielleicht etwas knapp.
- Nur die PRO Version bietet, so wie ich das sehe,  LAeq Werte d.h. den A-bewerteten äquivalenten Dauerschallpegel (oder die über den Zeitbereich gemittelte Energie).  Darauf beruhen alle Regelungen zum Lärm.

Mir ist nicht klar, was Du mit dem Rohr meinst? Meinst Du das Mikrofon im 1/2" Rohr? Könntest Du Deine Frage noch präzisieren.

Viele Grüße

Helmut

klacol, 2025-05-07:
Hallo Helmut,

danke für die Einschätzung. Ich habe mir schon gedacht, dass man am besten die PRO Version nutzen sollte.

Ja, ich meine das Mikrofon in dem Rohr. Ich hatte gar nicht verstanden, dass das Mikrofon in dem Rohr ist. Das wird vermutlich einen Grund in der Akustik oder Messtechnik haben, oder warum packt man das Mikrofon in ein Rohr?
Gruß Klaus

hbitter, 2025-05-07:
Hallo klacol,

bei Messmikrofonen ist es sehr gebräuchlich, dass das Mikrofon in einem Rohr mit dem Durchmesser von 1/2" sitzt.
- dieser Standard erlaubt die Verwendung von sogenannten Kalibratoren, die auf das 1/2" Rohr aufgesteckt werden, um das Mikrofon meist mit einem Sinus von 1000 Hz und einem Schalldruck von 94 dB (oder auch 114 dB) zu überprüfen. 
- Aufgrund der runden Form und des relativ kleinen Durchmessers ergeben sich auch akustische Vorteile. Es gibt weniger Reflexionen, die das Messergebnis verfälschen. Es sollten insgesamt vom Gehäuse möglichst wenig Reflexionen erzeugt werden, deshalb Rohr und konische Formen bei den Messmikrofonen.
Aus diesen Gründen haben wir ein Rohr mit 1/2" Durchmesser gewählt. Die Korrektur des Frequenzgangs bezieht sich nicht nur auf das Mikrofon selbst sondern auch auf den Einbau und das Gehäuse.

Viele Grüße
Helmut